litex-hub/linux-on-litex-vexriscv #261: ac701 board no eth0 interface

cyntem, 2021-12-28:
Hi, big thanks for great project!
I'm trying to get linux with ethernet and sdcard on ac701 board. 
My problems with this board was:
1) The board has low voltage on USB-UART CP2103GM. VIO port has 1.7V, but 1.8V is minimal for CP2103GM.  I don't know is it problem with my board or design mistake, but similar KC705 use CP2103GM with 2.5V, thats why KC705 has no such problem. I use external USB-UART converter with J48 3.3V header and serial interface become to work.
2) Another problem with DDR3 memory. In file litex_boards/targets/xilinx_ac701.py  
`PHYPadsReducer(platform.request("ddram"), [0, 1, 2, 3]),`  always produce memory test errors.
For my board working sequence `PHYPadsReducer(platform.request("ddram"), [1, 2, 3, 4]),` . With current sequence linux start boot.
3) Last problem - is ethernet interface.  **TFTP boot is working**, but eth0 interface is missing. 
I got errors 
`[    1.980624] liteeth f0002000.mac: unable to get rx-fifo-depth
[    1.982475] liteeth: probe of f0002000.mac failed with error -22`

Is it problem with linux image or with board? 

full log is 
`        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 28 2021 20:15:20
 BIOS CRC passed (1334d4e5)

 Migen git sha1: ac70301
 LiteX git sha1: a6ed4c5c

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
SDRAM:		524288KiB 32-bit @ 800MT/s (CL-7 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:6 m2:6 m3:6 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111000000000000000000000| delays: 05+-05
  m0, b04: |00000000000000011111111111100000| delays: 21+-06
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 21+-05
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |11111111100000000000000000000000| delays: 04+-04
  m1, b04: |00000000000000111111111110000000| delays: 19+-05
  m1, b05: |00000000000000000000000000000001| delays: 31+-00
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b04 delays: 19+-05
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |11111111111100000000000000000000| delays: 06+-06
  m2, b04: |00000000000000011111111111110000| delays: 21+-06
  m2, b05: |00000000000000000000000000000001| delays: 31+-00
  m2, b06: |00000000000000000000000000000000| delays: -
[... 137 lines cut ...]
[    1.986026] i2c /dev entries driver
[    2.006144] litex-mmc f0006000.mmc: Requested clk_freq=12500000: set to 12500000 via div=8
[    2.045537] NET: Registered protocol family 10
[    2.055988] Segment Routing with IPv6
[    2.057809] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    2.067912] NET: Registered protocol family 17
[    2.079082] Freeing unused kernel memory: 208K
[    2.080167] Kernel memory protection not selected by kernel config.
[    2.082153] Run /init as init process
[    2.084144] litex-mmc f0006000.mmc: Requested clk_freq=0: set to 390625 via div=256
[    2.106411] litex-mmc f0006000.mmc: Requested clk_freq=12500000: set to 12500000 via div=8
[    2.147306] litex-mmc f0006000.mmc: Requested clk_freq=25000000: set to 25000000 via div=4
[    2.150296] mmc0: new SDXC card at address 0001
[    2.158874] mmcblk0: mmc0:0001 SD64G 58.2 GiB 
[    2.169367]  mmcblk0: p1
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [    3.220688] random: dd: uninitialized urandom read (512 bytes read)
OK
Starting network: OK

Welcome to Buildroot
buildroot login: 
`

cyntem, 2021-12-29:
Finally i got eth0 worked!
Prebuilt linux images produce current bug (  linux_2020_12_30.zip and linux_2021_03_29 ).
I built new linux image and get it work. 

For now linux image from this man is not work.
```
$ git clone http://github.com/buildroot/buildroot
$ cd buildroot
$ make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
$ make
```
It's failed because of  headers 5.15.

I use buildroot **https://github.com/buildroot/buildroot/archive/refs/tags/2021.02.3.tar.gz** And built successfully.
Thanks for great project!

closed 2021-12-29 by cyntem

Dolu1990, 2022-11-18:
Hi,
Could you show the dts which was generated (in build/??? folder) ? Maybe it is missing there ?

cyntem, 2022-11-19:
Hi @smosanu ,
My last work version was with kernels 5.14, with 5.15 it's not worked. You can make menuconfig and choose custom 5.14 kernel to build image. For me it's work.  Also AC701 has two ethernet interface A7_1000BASEX and RGMII. Only RGMII work's stable on the board. I spent ours to understand what's wrong with 1000BASEX, but it's not works stable. Sometimes works, sometimes not, may be my board has a bug, I don't know.

smosanu, 2022-11-21:
@cyntem Thank you, I will attempt to use the 5.14 kernels, though several suggestions are to investigate the device tree (dts) file so I plan to compare with a known-to-work platform such as the Arty A7.
@Dolu1990 Here is the dts file and a log file.
[netboot_eth0_missing_log.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10053563/netboot_eth0_missing_log.txt)
[kc705.dts.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10053564/kc705.dts.txt)

Dolu1990, 2022-11-21:
>  I should probably enable it somehow?

I don't know how to do it either ^^

the dts seems fine to me. But weird there is realy nothing is the boot logs.
Can you send the linux .config from the linux repo  ?

smosanu, 2022-12-06:
Hi @Dolu1990 sorry for the late reply. I experimented with several buildroot configurations, all based on the original one from this repository. The boot log previously attached was with the original image downloaded from this repo, though it was as unsuccessful with images I generated with variations of this config.
[litex_vexriscv_defconfig.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10169999/litex_vexriscv_defconfig.txt)
I added (not all at once) lines 21, 44-46 (for benchmarks), 60. Will update here if I find a solution.

Dolu1990, 2022-12-06:
@smosanu Ahh i mean the linux defconfig, not the buildroot one.

Dolu1990, 2022-12-07:
more precisly, the .config from the output/build/linux-xxxxx folder :)

smosanu, 2022-12-07:
Here it is! I am uploading several - the file name describes the differences. Thank you for the precise instructions - I did need them. :)
[config_default.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10178565/config_default.txt) - this is the default configuration
[config_withDhryStone.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10178584/config_withDhryStone.txt)
[config_Dhrystn_plusEthAttempt.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10178592/config_Dhrystn_plusEthAttempt.txt)
[config_EthAttempt2.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10178600/config_EthAttempt2.txt)
[config_with_buildroot2021_02_3.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10178613/config_with_buildroot2021_02_3.txt)