litex-hub/linux-on-litex-vexriscv #269: dts generation error on arty board

controlpaths, 2022-03-24:
hello, 

I am trying to run the project in an arty board. When I execute make.py --board arty --cpu-count 1 --build i have the next error.

`
Error: build/arty/arty.dts:108.14-15 syntax error
FATAL ERROR: Unable to parse input tree
`

When I check the /build/arty.dts file I found that there is a mistake in the line 108.

          soc {{
            #address-cells = <1>;
            #size-cells    = <1>;
            compatible = "simple-bus";
            interrupt-parent = <&intc0>;
            ranges;

            soc_ctrl0: soc_controller@f0000000 {
                compatible = "litex,soc-controller";
                reg = <0xf0000000 0xc>;
                status = "okay";
            };

There is a double { symbol.

I think that I am doing something wrong but I can't find out what.

Thanks for your help!

Dolu1990, 2022-03-25:
Hi,

Do you have a old version of litex ? Older than https://github.com/enjoy-digital/litex/commit/f4967cfb85500e0a2736ce2ab7678271390ef715 ?

controlpaths, 2022-03-25:
Hi! thanks! yes I didn't realize that I am not in head.

closed 2022-03-25 by controlpaths