litex-hub/linux-on-litex-vexriscv #280: Executable for Linux

Ankelih, 2022-05-30:
Hello everyone, 

I want to run a c code on linux on litex vexriscv. Howevere there isn't gcc compiler. So is there a way to add a executable to image so system will be built with this executable and I can run this on linux?

Thanks.

Dolu1990, 2022-05-31:
i guess if you take let's say the sifive gcc : 

```
version=riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14
wget -O riscv64-unknown-elf-gcc.tar.gz riscv https://static.dev.sifive.com/dev-tools/$version.tar.gz
tar -xzvf riscv64-unknown-elf-gcc.tar.gz
sudo mv $version /opt/riscv
echo 'export PATH=/opt/riscv/bin:$PATH' >> ~/.bashrc
```

you write a main, and do a 
compile it with -march=rv32ima -mabi=ilp32 it may be good to go for the compilation side of things.

To add things to the image, either you add them manualy, either you add things to the overlay : https://github.com/litex-hub/linux-on-litex-vexriscv/tree/master/buildroot/board/litex_vexriscv/rootfs_overlay

And build buildroot.

Else another way would be to define a buildroot package. Ex : 
https://github.com/litex-hub/linux-on-litex-vexriscv/tree/master/buildroot/package/dhrystone-opt

But more like https://github.com/SpinalHDL/buildroot-spinal-saxon/tree/main/package/mandelbrot

ashuthosh-mr, 2022-08-17:
I tried compiling with -march=rv32ima -mabi=ilp32 a simple hello world program. I uploaded the newly built buildroot. But while running the executable I get illegal instructions. I tried running uart.elf from briey soc repository, then I got Segmentation fault. Is there documentation, sample executable that can be run?

ashuthosh-mr, 2022-08-18:
It's working when I compile using riscv64 sifive toolchain. Thank you.