litex-hub/linux-on-litex-vexriscv #282: how to build and create opensbi.bin and image files as in Pre-built Bitstreams and Linux/OpenSBI images

zoro269, 2022-06-06:
https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164#issue-767463348 
I have followed the steps but still not working. 
The purpose I want is to be able to build an openSBI image file like in the instructions above

zoro269, 2022-06-07:
> Hi @zoro269 <https://github.com/zoro269>,
>
> here is some code I used recently and that will probably be integrated
> soon in the repository:
>
> def opensbi_build():
>     # Be sure third_party dir is present and switch to it.
>     create_third_party_dir()
>     os.chdir("third_party")
>
>     # Get OpenSBI.
>     if not os.path.exists("opensbi"):
>         os.system("git clone https://github.com/litex-hub/opensbi --branch 0.8-linux-on-litex-vexriscv")
>     os.chdir("opensbi")
>
>     # Configure / Build OpenSBI.
>     os.system("make CROSS_COMPILE=riscv64-unknown-elf- PLATFORM=litex/vexriscv")
>
>     # Copy Linux Images.
>     os.system("cp build/platform/litex/vexriscv/firmware/fw_payload.bin ../../images/opensbi.bin")
>
> Please let me know if you still have troubles building with this.

Hi enjoy-digital
Can you guide me the running steps and order of running? I still don't
quite understand.
Many thanks

enjoy-digital, 2022-06-07:
This should work:
````
git clone https://github.com/litex-hub/opensbi --branch 0.8-linux-on-litex-vexriscv
cd opensbi
make CROSS_COMPILE=riscv64-unknown-elf- PLATFORM=litex/vexriscv
````
The binary is then located at `build/platform/litex/vexriscv/firmware/fw_payload.bin`.

zoro269, 2022-06-07:
I followed the instructions and got some errors as follows:
- first: at make CROSS_COMPILE=riscv64-unknown-elf- PLATFORM=litex/vexriscv . I got the following error:

 ELF       platform/litex/vexriscv/firmware/fw_dynamic.elf
/home/namnguyendc4/Testtt/linux-on-litex-vexriscv/opensbi/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/8.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/namnguyendc4/Testtt/linux-on-litex-vexriscv/opensbi/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/8.1.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/home/namnguyendc4/Testtt/linux-on-litex-vexriscv/opensbi/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/8.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /home/namnguyendc4/Testtt/linux-on-litex-vexriscv/opensbi/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/8.1.0/libgcc.a(div.o)
collect2: error: ld returned 1 exit status
make: *** [Makefile:317: /home/namnguyendc4/Testtt/linux-on-litex-vexriscv/opensbi/build/platform/litex/vexriscv/firmware/fw_dynamic.elf] Error 1

zoro269, 2022-06-07:
This leads to the file opensbi/build/platform/litex/vexriscv/firmware there is no file fw_payload.bin
and here is a picture of my firmware
[image]

enjoy-digital, 2022-06-07:
@zoro269: Can you test with the GCC toolchain provided by `python3 litex_setup.py --gcc=riscv`?

zoro269, 2022-06-07:
@enjoy-digital I have installed and tried the toolchain and still have the error

zoro269, 2022-06-08:
> @zoro269: Can you test with the GCC toolchain provided by `python3 litex_setup.py --gcc=riscv`?
> 
> (https://github.com/enjoy-digital/litex/blob/master/litex_setup.py#L323-L345)

@enjoy-digital 
i tried it and it still fails
[image]

zoro269, 2022-06-20:
@enjoy-digital 

> @zoro269: Can you test with the GCC toolchain provided by `python3 litex_setup.py --gcc=riscv`?
> 
> (https://github.com/enjoy-digital/litex/blob/master/litex_setup.py#L323-L345)

Thanks guys, I fixed it with the gcc toolchain that was shown and downloaded in the litex_setup.py file. But can you guys explain to me the difference between these two versions of gcc toolchain?

enjoy-digital, 2022-07-07:
Hi @zoro269,

the OpenSBI build probably require a minimum GCC version to build correctly. We haven't analyzed this so just recommend using the version that has been used successfully.

myy1966, 2024-12-27:
make CROSS_COMPILE=riscv64-unknown-elf- PLATFORM=litex/vexriscv
I get this error:

Dolu1990, 2024-12-27:
Either you get a older version of gcc (11.x) either would need to fix the flow to add the _zicsr to the march

myy1966, 2024-12-30:
> Either you get a older version of gcc (11.x) either would need to fix the flow to add the _zicsr to the march

After changed PLATFORM_RISCV_ISA from rv32ima to rv32ima_zicsr_zifencei,  I successfully built 1.3.1-linux-on-litex-vexriscv branch opensbi binary and used it to run sim.py.

I need to highlight two points that differ from enjoy-digital's description:
1. After changed PLATFORM_RISCV_ISA from rv32ima to rv32ima_zicsr_zifencei, 0.8-linux-on-litex-vexriscv can built to binary but failed to boot linux. We should use 1.3.1-linux-on-litex-vexriscv.
2. We should use fw_jump.bin instead of fw_payload.bin. The fw_payload.bin file has reached a size of 1.1 GB and contains a dump payload instead of booting the user-provided Linux kernel image.