litex-hub/linux-on-litex-vexriscv #287: SPI SD card boot failing on QMTech 5CEFA2

fritzbauer, 2022-06-15:
First, thank you so much for working on litex! It is an amazing help when learning about SoCs, FPGAs and how to approach building such systems.

This is not an issue I urgently need to resolve for myself, but I want to report it so others do not face it in the future. I am willing to help with the investigation, but need some guidance on it.

### Prerequisites
I was able to build and serialboot on the QMTech Cyclone V 5CEFA2 Board after adding it to make.py like this: https://github.com/litex-hub/linux-on-litex-vexriscv/pull/286 
As a second step, I added an `"spisdcard"` to the [soc_capabilities in make.py](https://github.com/litex-hub/linux-on-litex-vexriscv/pull/286/files#diff-6bf738d0dd0f963c9f86b7917b2513277e7b3f78eeea507f809f49bdcad61733R598) and added the Pins in [litex_boards/platforms/qmtech_5cefa2.py](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/qmtech_5cefa2.py):
```
("spisdcard", 0,
  Subsignal("cs_n", Pins("J3:18")),
  Subsignal("clk", Pins("J3:20")),
  Subsignal("mosi", Pins("J3:16")),
  Subsignal("miso", Pins("J3:22")),
  IOStandard("3.3-V LVTTL"),
),
```

During investigation I also ended up enabling the software_debug flag on the [litex_boards/targets/qmtech_5cefa2.py](
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/qmtech_5cefa2.py#L171) by setting `soc.add_spi_sdcard(software_debug=True)`.

### Issue
Attempting to boot from the SD card is failing and it seems like the SoC just crashes. At least it does not respond on the UART anymore.

The output is:
```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
S
```

I formatted my SD card with a FAT-16 partition and I am able to mount, read and write on this SD card once I booted to linux. (in order to boot I had to [uncomment the call to sdcardboot()](https://github.com/enjoy-digital/litex/blob/d537e30c5a0688461a0dbbef7bfb132dd4cc5618/litex/soc/software/bios/main.c#L65)

### Not using linux-on-vexriscv
I was able to get the SD card to load when just using a plain build of litex-boards:
```
python litex_boards/targets/qmtech_5cefa2.py --cpu-type=vexriscv --cpu-variant=standard+debug --build
```
The output of this working boot is:
```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
Setting SDCard clk freq to 410 KHz
Setting SDCard clk freq to 17 MHz
Copying Image to 0x40000000 (7531468 bytes)...
[########################################]
Copying rv32.dtb to 0x40ef0000 (3185 bytes)...
[########################################]
Copying rootfs.cpio to 0x41000000 (3781632 bytes)...
[########################################]
Copying opensbi.bin to 0x40f00000 (53640 bytes)...
[########################################]
Executing booted program at 0x40f00000

--============= Liftoff! ===============--
```
It ends up in a trap when lifting off, but I reckon this is just due to the fact that the plain litex-board is lacking some capabilities required to run linux.

### Ideas

- Could this issue be caused by the small size of SRAM and L2? It is 2KiB each, while the plain litex-board has 8KiB each. Unfortunately quartus is not able to fit larger sram on the chip.
  - If this is the case it might be nice to somehow tell the bios to use the SDRAM instead.
- I tried adding additional debug logging to the bios, however it is very difficult, because the UART just crashes at some point. But from the log output above you can see that it at least starts outputting the line "Setting SDCard clk freq to 410 KHz" but fails after sending the "S". But this character is not always sent.
- After failing to identify the cause with printf's I worked through the OpenOCD, GDB tutorial. I added a second serial_debug uart to the board definition on two free IO Pins. I was able to get it working on a plain litex-board. The gdb connection is not very stable, it oftentimes just starts hanging when trying to "step". But telnet to openocd allows me to halt, step and resume the bios, though it is not possible to add breakpoints. With all that knowledge, I wanted to start debugging the linux-on-vexriscv SoC, however I was not able to figure out how to use the vexriscv linux+debug core.
  - If setting up openocd is the right track to nail this issue down, could you please let me know, if it will be possible to add the debugging ability without too much effort?

Please find the build output and the quartus reports attached:
[spisdcard_linux-on-vexrisc_boot_on_5CEFA2.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/8911369/spisdcard_linux-on-vexrisc_boot_on_5CEFA2.zip)

Thank you for your help!

fritzbauer, 2022-06-16:
Good news.
I found out the issue consists of two issues:
* A stackoverflow when attempting to decrement the stackpointer by 2035 when entering [sdcardboot_from_json](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/boot.c#L714)
* The spisdcard code not initializing the sdcard properly. It only starts working in the bios once I booted linux once

fritzbauer, 2022-06-23:
Reported the stackoverflow issue in the litex repo now: https://github.com/enjoy-digital/litex/issues/1339
My current workaround is to adjust bios/linker.ld to place _stack in the main_ram.

fritzbauer, 2022-06-23:
Was able to fix the initialization issue by skipping the select functions for CMD0:
https://github.com/enjoy-digital/litex/pull/1340

So the last open question for the linux-on-litex-vexriscv project:
Would it be possible to add vexriscv openocd debugger build options to simplify troubleshooting in the future?

enjoy-digital, 2022-07-07:
Thanks @fritzbauer, the SDCard issue has been fixed and I'll have a closer look at the stackoverflow issue.

For OpenOCD, we could enable it in the future, I'll first see this in LiteX and this should also allow to simply add it here.