litex-hub/linux-on-litex-vexriscv #288: USB host support

machdyne, 2022-06-22:
I have designed an FPGA-based computer (Schoko) that runs Linux and has video output thanks to LiteX and this project. The computer has a USB host port wired directly to the FPGA, with 15K pull-downs. I know that LiteX has a USB OHCI wrapper module and apparently at least one project has accomplished USB keyboard/mouse support in Linux using LiteX, but I don't believe there is an open-source solution available yet.

In short, it would be great if linux-on-litex had USB host support.

Dolu1990, 2022-06-22:
> but I don't believe there is an open-source solution available yet.

Do you mean https://github.com/litex-hub/pythondata-misc-usb_ohci ? 
If yes, it is opensource :
- https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Libraries/Com/usb_ohci.html?highlight=ohci
- https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/com/usb/ohci/UsbOhciWishbone.scala
- https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/com/usb/ohci/UsbOhci.scala

Also, got usb flash drive / hub / headphone to work too on it :)

Dolu1990, 2022-06-23:
Maybe, one good reference is : 
https://github.com/enjoy-digital/litex_vexriscv_smp_usb_host_test

See [digilent_arty.py](https://github.com/enjoy-digital/litex_vexriscv_smp_usb_host_test/blob/master/digilent_arty.py)

And the few USB / OHCI addition in the buildroot config.

I guess the ohci controller isn't added automaticaly in the DTS, if it isn't added, let me know ^^

machdyne, 2022-06-24:
Everything looks good during boot, but when I plug in any USB device I see this:

```
[   64.121736] usb 1-1: new low-speed USB device number 2 using ohci-platform
[   64.404893] ohci-platform c0000000.mac: frame counter not updating; disabled
[   64.406877] ohci-platform c0000000.mac: HC died; cleaning up
[   69.210696] usb usb1-port1: attempt power cycle
```

Here's the USB boot messages:

```
[   16.650366] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[   16.653178] ohci-platform: OHCI generic platform driver
[   16.662612] ohci-platform c0000000.mac: Generic Platform OHCI controller
[   16.667152] ohci-platform c0000000.mac: new USB bus registered, assigned bus number 1
[   16.675406] ohci-platform c0000000.mac: irq 2, io mem 0xc0000000
[   16.793694] usb usb1: New USB device found, idVendor=1d6b, idProduct=0001, bcdDevice= 5.18
[   16.797532] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[   16.800982] usb usb1: Product: Generic Platform OHCI controller
[   16.802973] usb usb1: Manufacturer: Linux 5.18.0-rc7 ohci_hcd
[   16.806136] usb usb1: SerialNumber: c0000000.mac
[   16.870986] hub 1-0:1.0: USB hub found
[   16.878133] hub 1-0:1.0: 1 port detected
[   16.931134] usbcore: registered new interface driver usb-storage
[   16.945539] usbcore: registered new interface driver ftdi_sio
[   16.950850] usbserial: USB Serial support registered for FTDI USB Serial Device
[   16.962574] UDC core: couldn't find an available UDC - added [g_audio] to list of pending drivers
[   17.001481] mousedev: PS/2 mouse device common for all mice
[   17.014230] i2c_dev: i2c /dev entries driver
[   17.119440] usbcore: registered new interface driver usbhid
[   17.122247] usbhid: USB HID core driver
[   17.155464] usbcore: registered new interface driver snd-usb-audio
```

Dolu1990, 2022-06-24:
Hmmm maybe there is something wrong on the d+ d- ?
Could you take a probe of it ? To see if it stuck in a non-idle mode ?
Also, could you cat /proc/interrupts to see if you get some interrupts from the controller ?

machdyne, 2022-06-24:
```
root@buildroot:~# cat /proc/interrupts 
           CPU0       
  2:          1  SiFive PLIC  16 Edge      ohci_hcd:usb1
  5:    3382729  RISC-V INTC   5 Edge      riscv-timer
IPI0:         0  Rescheduling interrupts
IPI1:         0  Function call interrupts
IPI2:         0  CPU stop interrupts
IPI3:         0  IRQ work interrupts
IPI4:         0  Timer broadcast interrupts
```

machdyne, 2022-06-24:
Immediately after booting there are 0 interrupts. After plugging in a USB keyboard there are now 2 interrupts. Plugging in the keyboard a second time has no effect.

I will probe the hardware and report back.

The schematics are available here if you'd like to have a look:
https://github.com/machdyne/schoko/blob/main/pcb/schoko_v1.pdf

Thanks.

machdyne, 2022-06-24:
It's working now. The problem was that I had timing violations, I reduced the clock speed and now everything is great.

I guess linux-on-litex does have USB host support. I will plan to write some documentation along with adding support for my board and submit a PR in the future.

Thanks again for your help.

Dolu1990, 2022-06-24:
> It's working now. The problem was that I had timing violations, I reduced the clock speed and now everything is great.

Hooo nice catch :D

>  I will plan to write some documentation along with adding support for my board and submit a PR in the future.

Thanks :)

enjoy-digital, 2022-07-07:
Thanks @machdyne, @enjoy-digital, we should definitely integrate the USB-Host support directly in this repository. I'll do this soon.

commit 0a709af7 (2022-07-21): Merge pull request #293 from machdyne/master

add schoko support and usb host configs resolves #288
closed 2022-07-21 by enjoy-digital