litex-hub/linux-on-litex-vexriscv #289: sdcard boot failed on the kcu105 board

davidgaow, 2022-06-29:
Hi LiteX,   on the xilinx board kcu105, sdcard boot failed,  and the  command `sdcard_freq 40000000` would give the result of 31MHz !?

`litex> sdcard_detect

SDCard inserted.

litex> sdcard_freq

sdcard_freq <freq>
litex> sdcard_freq 40000000

Setting SDCard clk freq to 31 MHz

litex> sdcard_init

Initialize SDCard... Successful.

litex> sdcardboot

Booting from SDCard in SD-Mode...
Booting from boot.json...
ff.c, func: f_mount:   vol = 0
ff.c, func: f_mount:  fs exist
ff.c, func: f_mount:  res = mount_volume(&path, &fs, 0); res=13
Booting from boot.bin...
ff.c, func: f_mount:   vol = 0
ff.c, func: f_mount:  fs exist
ff.c, func: f_mount:  res = mount_volume(&path, &fs, 0); res=13
SDCard boot failed.
`

davidgaow, 2022-06-29:
litex> reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Oct 29 2021 17:11:59
 BIOS CRC passed (555d7a39)

 Migen git sha1: 27dbf03
 LiteX git sha1: 78c1751c

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 125MHz
BUS:		AXI-LITE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		2KiB
SDRAM:		1048576KiB 64-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 424
  Cmd/Clk scan (0-212)
  |1111  |101111111  |101110111  |000001111| best: 120
  Setting Cmd/Clk delay to 120 taps.
  Data scan:
  m0: |11100000000000001111111111| delay: 243
  m1: |11110000000000000111111111| delay: 261
  m2: |11111100000000000001111111| delay: 295
  m3: |11111110000000000000111111| delay: 00
  m4: |11111110000000000000111111| delay: -
  m5: |11111110000000000000111111| delay: 00
  m6: |11111111100000000000001111| delay: 00
  m7: |11111111100000000000001111| delay: 00
Write latency calibration:
m0:0 m1:0 m2:0 m3:6 m4:6 m5:6 m6:6 m7:6 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111000000000000000000000000000| delays: 34+-34
  m0, b03: |00000011111111111100000000000000| delays: 184+-93
  m0, b04: |00000000000000000000111111111110| delays: 401+-89
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b03 delays: 183+-95
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |11111000000000000000000000000000| delays: 34+-34
  m1, b03: |00000001111111111100000000000000| delays: 189+-90
  m1, b04: |00000000000000000000111111111110| delays: 402+-91
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 187+-90
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |11100000000000000000000000000000| delays: 20+-20
  m2, b03: |00000111111111111000000000000000| delays: 164+-88
  m2, b04: |00000000000000000011111111111100| delays: 375+-91
  m2, b05: |00000000000000000000000000000000| delays: 00+-08
[... 50 lines cut ...]
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (2.0MiB)...
  Write speed: 39.7MiB/s
   Read speed: 21.8MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SD-Mode...
Booting from boot.json...
ff.c, func: f_mount:   vol = 0
ff.c, func: f_mount:  fs exist
ff.c, func: f_mount:  res = mount_volume(&path, &fs, 0); res=13
Booting from boot.bin...
ff.c, func: f_mount:   vol = 0
ff.c, func: f_mount:  fs exist
ff.c, func: f_mount:  res = mount_volume(&path, &fs, 0); res=13
SDCard boot failed.
No boot medium found

--============= Console ================--

AEW2015, 2022-08-17:
This is a specific issue with the KU105 as a ZYNQ system manager on the board.
Review these comments in a previous issue for a solution:
https://github.com/litex-hub/linux-on-litex-vexriscv/issues/238#issuecomment-910810539