litex-hub/linux-on-litex-vexriscv #290: Buildroot with x11 / doom / dhrystone config

Dolu1990, 2022-07-01:
Here is an achive with the buildroot config for x11 + chocolate-doom + dhrystone-opt

[buildroot_x11_doom.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/9030378/buildroot_x11_doom.zip)

Note you need to get buildroot with the git hash 9ef54b7d

pdlloyd, 2022-08-15:
@Dolu1990 this is really cool and something I've been trying to get working on my ULX3S. Unfortunately, I'm a bit inexperienced with these tools and was wondering if you could provide some clarification about why buildroot needs to be at that git hash, and how to be sure the dts ramfs is configured with enough space to eat the whole cpio. Thanks!

Dolu1990, 2022-08-17:
No worries :)

> if you could provide some clarification about why buildroot needs to be at that git hash

So, if i remember well, it was because they removed SDL2 support for 32 bits risc-v, something in that kind, or compilation issue.

Also in general, there is so many thing which could break in the whole process, i personnaly always try to stay on a functionnal version of the things, to keep my sanity at a managable level XD

> and how to be sure the dts ramfs is configured with enough space to eat the whole cpio. Thanks!

So by default the dts ramfs is configured to 8 GB, which is too low for x11 stuff.
So, either you need to manualy check the dts in the build/xxx folder, check that the range between : 
        linux,initrd-start = <xxx>;
        linux,initrd-end = <yyy>;
is bigger than the cpio file size, and then generate the DTB manuals out of the DTS.

Either you try to switch to using the sdcard to host a second partition with the buildroot file system, this quite better for low memory systems, as ULX3s.

smosanu, 2023-01-11:
Particularly useful is the `BR2_TOOLCHAIN_BUILDROOT_CXX=y` line which enables the generation of the riscv-buildroot-linux-g++ compiler

Basman74, 2024-10-02:
Hello! Apologies for bringing up an old thread. Merely curious to know what specific toolchain setup (gcc/glibc versions, etc.) is necessary to compile buildroot hash 9ef54b7d? For sure my existing build environment does _not_ do it. 
Thanks in advance!