litex-hub/linux-on-litex-vexriscv #291: C++ execution code error

RemyBLT, 2022-07-07:
Hello LiteX community,

I'm somehow stuck for a month now with a problem of code execution. I built my own riscv toolchain with the Linux-with-Litex with the VexRiscV as a target of my C++ code. As I need to use libraries/peripherals from the embedded Linux, I firstly tried to build the Linux version of the Toolchain and tried a simple "Hello World" using target's iostream library. But I get a weird output error saying that it has an "unhandled signal 7 code 0x1 at 0x95bca894 in ld-linux-riscv32-ilp32.so.1" as you can read in the attached picture.

[image]

I generated my toolchain using this git project : https://github.com/riscv-collab/riscv-gnu-toolchain
I used "./configure --prefix=/opt/riscv --enable-multilib --with-arch=rv32imafdc -with-abi=ilp32" and "make linux" to generate the tool "riscv32-unknown-linux-gnu-" because from my understanding, it is the one that we must use to use libraries from the target like "sys/mman.h". When I try to execute it with my own "riscv32-unknown-elf-" generated with the parameters set above, it works no matter if its C or C++. It's only an issue when I try to execute it with "riscv32-unknown-linux-gnu-". 

Did you ever got this error ? Do you know if there is any way to solve it ? Unfortunately, I can't see where I'm wrong...

Thank you very much

RemyBLT, 2022-07-07:
From my initial research, the signal 7 is, by the POSIX standard definition a SIGBUS signal which makes sense since Linux tells that this is a Bus Error. My guess is that the program somehow allocate memory outside of the range allocated by I can't find out why and if it's the case it's most certainly not a Toolchain parameter error

RemyBLT, 2022-07-18:
Okay so I've found a workaround that makes everything works. To compile the program, simply use the toolchain generated through the buildroot compilation that you can find in {buildroot dir}/output/host/bin. I used riscv32-linux-g++ with the parameter that targets the sysroot directory. To sum up, here is an example to compile a C++ program that uses shared libraries :
 
`{buildroot dir}/output/host/bin/riscv32-linux-g++ _program that you want to compile_ -o _output_filename_`

smosanu, 2023-01-11:
This issue was very useful, thank you! On my end, I added the `BR2_TOOLCHAIN_BUILDROOT_CXX=y` line to the config file to obtain the riscv32-buildroot-linux-g++ compiler tool in the output/host/bin/ and that seems to work.