litex-hub/linux-on-litex-vexriscv #295: Build for de0nano board

fdu, 2022-07-29:
I am trying to run the build for the de0nano board with

`./make.py --board=de0nano --cpu-count=1 --build`

This leads to build errors:

1. `Verilog HDL Unsupported Feature error at ***.v(14192): system function "$urandom" is not supported for synthesis`
2. If I remove the sections with $urandom then another error comes up:
```
Error (170011): Design contains 46196 blocks of type combinational node.  However, the device contains only 22320 blocks.
...
Error (171000): Can't fit design in device
```

As de0nano is listed in the supported boards, I expected the build to complete without issues. Am I using it wrong? Is there a workaround for those issues?

Thank you!

tpwrules, 2022-08-01:
This is solved by https://github.com/SpinalHDL/VexRiscv/issues/235#issuecomment-1035049599 . Unfortunately I'm not sure how to convince LiteX to add that option.

Dolu1990, 2022-08-03:
for "Design contains 46196 blocks of type combinational node" What configuration did you used ? the default one ? seems like some memories didn't inferred in blockram ?

fdu, 2022-08-04:
@tpwrules Where should SYNTHESIS be defined and with what value? Alternatively is it safe to simply remove the code under `ifndef SYNTHESIS ... endif`?

@Dolu1990 This was using litex_setup from branch 2022.04, linux-on-litex-vexriscv 0a709af74e668deb66e356cc3441d06467985467 and default configuration with command: `./make.py --board=de0nano --cpu-count=1 --build`

Let me know if you need more information on the environment.

Thanks a lot for your help.

fdu, 2022-08-05:
Latest master includes https://github.com/enjoy-digital/litex/pull/1392 which solves the SYNTHESIS issue.

Unfortunately when building with `--board=de0nano`, the other issue is still present: `Design contains 46196 blocks of type combinational node.  However, the device contains only 22320 blocks.`. With `--board=de10nano` the build completes without error. Any idea what the problem could be?

jeremy-heath, 2022-08-11:
reduce "integrated_sram_size"
class De0Nano(Board):
    soc_kwargs = {"l2_size" : 2048, "integrated_sram_size" : 0x800,} # Use Wishbone and L2 for memory accesses.

PR #298 (2022-09-05): make.py: Reduce integrated_sram_size for de0nano
This reduce integrated_sram_size to 0x800 to fix #295.
commit e5cdcf95 (2022-09-05): make.py: Reduce integrated_sram_size for de10nano\

This reduce integrated_sram_size to 0x800 to fix #295.
closed 2022-09-05 by enjoy-digital