litex-hub/linux-on-litex-vexriscv #296: Build failed for Arty-A7 Vivado: ERROR: [DRC UTLZ-1] Resource utilization: XADC over-utilized in Top Level Design

mrinformatics, 2022-08-09:
An error occurred during the fresh compilation for the arty_a7 board using the command
--> ./make.py --board arty_a7 --cpu-count=1 --build

Screenshot attached: -
[image]

AEW2015, 2022-08-16:
I got this too with the arty and arty_a7.

There are two xadc primtives:
```
XADC #(
	.INIT_40(16'd36864),
	.INIT_41(14'd12016),
	.INIT_42(11'd1024),
	.INIT_48(15'd18177),
	.INIT_49(4'd15),
	.INIT_4A(15'd18176),
	.INIT_4B(1'd0),
	.INIT_4C(1'd0),
	.INIT_4D(1'd0),
	.INIT_4E(1'd0),
	.INIT_4F(1'd0),
	.INIT_50(16'd46573),
	.INIT_51(15'd22937),
	.INIT_52(16'd41287),
	.INIT_53(16'd56797),
	.INIT_54(16'd43322),
	.INIT_55(15'd20753),
	.INIT_56(16'd37355),
	.INIT_57(16'd44622),
	.INIT_58(15'd22937),
	.INIT_5C(15'd20753)
) XADC (
	.CONVST(1'd0),
	.CONVSTCLK(1'd0),
	.DADDR(main_xadc_dadr),
	.DCLK(sys_clk),
	.DEN(main_xadc_den),
	.DI(main_xadc_di),
	.DWE(main_xadc_dwe),
	.RESET(sys_rst),
	.VAUXN(1'd0),
	.VAUXP(1'd0),
	.VN(1'd0),
	.VP(1'd0),
	.ALM(main_xadc_alarm),
	.BUSY(main_xadc_busy),
	.CHANNEL(main_xadc_channel),
	.DO(main_xadc_do),
	.DRDY(main_xadc_drdy),
	.EOC(main_xadc_eoc),
	.EOS(main_xadc_eos),
	.OT(main_xadc_ot)
);
```

```
XADC #(
	.INIT_40(16'd36864),
	.INIT_41(14'd12016),
	.INIT_42(11'd1024),
	.INIT_48(15'd18177),
	.INIT_49(4'd15),
	.INIT_4A(15'd18176),
	.INIT_4B(1'd0),
	.INIT_4C(1'd0),
	.INIT_4D(1'd0),
	.INIT_4E(1'd0),
	.INIT_4F(1'd0),
	.INIT_50(16'd46573),
	.INIT_51(15'd22937),
	.INIT_52(16'd41287),
	.INIT_53(16'd56797),
	.INIT_54(16'd43322),
	.INIT_55(15'd20753),
	.INIT_56(16'd37355),
	.INIT_57(16'd44622),
	.INIT_58(15'd22937),
	.INIT_5C(15'd20753)
) XADC_1 (
	.CONVST(1'd0),
	.CONVSTCLK(1'd0),
	.DADDR(soclinux_dadr),
	.DCLK(sys_clk),
	.DEN(soclinux_den),
	.DI(soclinux_di),
	.DWE(soclinux_dwe),
	.RESET(sys_rst),
	.VAUXN(1'd0),
	.VAUXP(1'd0),
	.VN(1'd0),
	.VP(1'd0),
	.ALM(soclinux_alarm),
	.BUSY(soclinux_busy),
	.CHANNEL(soclinux_channel),
	.DO(soclinux_do),
	.DRDY(soclinux_drdy),
	.EOC(soclinux_eoc),
	.EOS(soclinux_eos),
	.OT(soclinux_ot)
);
```

AEW2015, 2022-08-17:
Comment out this line and is should work:
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L92

It looks like a recent update to litex boards added the xadc there, so we don't need to do it in the soc.

enjoy-digital, 2023-01-23:
Thanks  @thesaadmemon and @AEW2015. Fixed with https://github.com/litex-hub/litex-boards/commit/b8abdf1b39d322101e5ec88d0e0650b42f836a80.

closed 2023-01-23 by enjoy-digital