litex-hub/linux-on-litex-vexriscv #303: __sync_bool_compare_and_swap error while compilation

amr-25, 2022-08-25:
Hi,

I am compiling a c++ source using the command, riscv32-linux-g++ -march=rv32ima -mabi=ilp32 -std=c++11 -O3 -Wall -fopenmp src.c -o output. I get the following error posted below. One of the header files use _sync_bool_compare_and_swap which is causing the error. But it works fine when using g++ instead of riscv-linux-g++, Any thought on this? Thank you

/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function .L38': bfs.cc:(.text+0x2f0): undefined reference to __sync_bool_compare_and_swap_8'
/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function .L185': bfs.cc:(.text._ZN11BuilderBaseIiiiLb1EE7MakeCSRERK7pvectorI8EdgePairIiiEEbPPPiS8_._omp_fn.0[_ZN11BuilderBaseIiiiLb1EE9MakeGraphEv]+0xb0): undefined reference to __sync_fetch_and_add_8'
/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function .L170': bfs.cc:(.text._ZN11BuilderBaseIiiiLb1EE7MakeCSRERK7pvectorI8EdgePairIiiEEbPPPiS8_._omp_fn.0[_ZN11BuilderBaseIiiiLb1EE9MakeGraphEv]+0x104): undefined reference to __sync_fetch_and_add_8'
/home/user/Litex/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/10.4.0/../../../../riscv32-buildroot-linux-gnu/bin/ld: /tmp/ccYBJso0.o: in function .L184': bfs.cc:(.text._ZN11BuilderBaseIiiiLb1EE7MakeCSRERK7pvectorI8EdgePairIiiEEbPPPiS8_._omp_fn.0[_ZN11BuilderBaseIiiiLb1EE9MakeGraphEv]+0x15c): undefined reference to __sync_fetch_and_add_8'
collect2: error: ld returned 1 exit status

Dolu1990, 2022-08-25:
Hi,

Absolutly no idea, sorry

amr-25, 2022-08-30:
I tried using https://gcc.gnu.org/onlinedocs/gcc/_005f_005fatomic-Builtins.html instead of __sync builtins. The compilation problem solved but running the executable on the SMP gave segmentation fault with signal code 11. I might have missed something with the programming part(pointers and arrays). I didn't proceed and took another approach to write inline assembly for compare and swap, fetch and add, which then worked.

closed 2022-08-30 by amr-25