litex-hub/linux-on-litex-vexriscv #306: mpirun ./executable gives this message

amr-25, 2022-09-05:
Hi, I am trying to run MPI applications on the SMP. I have built buildroot using the BR2_PACKAGE_OPENMPI=y package since mpi.h was not recognised. It compiles fine without mpi wrapper by manually providng the include and library paths, flags. 
for example: 
./riscv32-linux-gcc -march=rv32ima -mabi=ilp32 -I /home/user/Litex/buildroot/output/host/riscv32-buildroot-linux-gnu/sysroot/usr/include/openmpi/ hello.c -Wl,-rpath -Wl,--enable-new-dtags -L /home/user/Litex/buildroot/output/host/riscv32-buildroot-linux-gnu/sysroot/usr/lib/openmpi/ -lmpi -lopen-rte -lopen-pal -lm -o hello

However on transfering to the board with rootfs.cpio, and running it either by ./hello or through mpirun ./hello, it gives an error:
[err] opal_libevent2022_evsig_init: Address family not supported by protocol.

I googled the error and it did not seem mpi related. Does anyone faced this issue or are aware of what this means?

Thank you

jeremy-heath, 2022-09-08:
Not really a linux-on-litex-vexriscv issue.  It is an openmpi issue.
Try
1:
Try anything
Let's suppose, this line defines your server name:
127.0.1.1 server-name
Change it as follows:
127.0.1.1 server-name.localhost server-name

2:
Continuing on our investigation, we found that the alternate way of disabling IPv6 support is to pass
ipv6.disable=1

as a bootarg to the Linux kernel

amr-25, 2022-09-08:
Thank you for the response. I did not understand the first one, how to do it. 

Regarding 1, The hosts listed in /etc/hosts are: 
127.0.0.1	localhost
127.0.1.1	buildroot

and /etc/hostname is:
buildroot

I tried the second, by using these commands to disable the ipv6.
sysctl -w net.ipv6.conf.all.disable_ipv6=1
sysctl -w net.ipv6.conf.default.disable_ipv6=1

But I still got the same error. Am I doing something wrong?

jeremy-heath, 2022-09-08:
Really recommend a more appropriate forum: open-mpi.org/community/lists/ompi.php

Try.
Make linux-menuconfig
    [*] Unix domain sockets

enjoy-digital, 2023-01-23:
As suggested by @jeremy-heath, the discussion seems related to OpenMPI package and probably not directly related to the current project. This could probably be discussed in OpenMPI forums.

closed 2023-01-23 by enjoy-digital