litex-hub/linux-on-litex-vexriscv #307: Send and receive through ethernet on arty a7

ashuthosh-mr, 2022-09-06:
Hi,

Sorry to have this question. This might be a simple question that I am struggling to solve for sometime. I want to quickly transfer the images and files through ethernet, and also receive the files generated by the vexriscv. I have programmed the arty7-35t fpga with the SMP. The serial boot is working fine. But since I have to debug the applications, it is slow to transfer via serial. I tried setting up TFTP server but was not able to transfer the images through ethernet. I referred the documentation (https://github.com/enjoy-digital/litex/wiki/Load-Application-Code-To-CPU). It would be a great help if there is already other documentation that might be helpful, I have looked at issue #65.

Thank you

smosanu, 2022-11-17:
Hi @ashuthosh-mr, were you able to get it to work? Were you able to see the `eth0` interface when you run `ifconfig -a`? I am struggling with a similar issue on the KC705 board. I was able to boot via TFTP so I know the ethernet interface works in BIOS but was not able to see it in Linux.

To get the TFTP working in BIOS make sure you set the host IP to 192.168.1.100 and have the TFTP server configured on your host machine with the image files in the correct directory.

ashuthosh-mr, 2022-11-20:
No. I wasn't able to get it work.

enjoy-digital, 2023-01-23:
Hi @ashuthosh-mr,

sorry for the delay, are you still trying to get this working? If so, I would recommend following @smosanu  advices and also testing the TFPServer manually first and when testing with the board, have Wireshark running to capture packets transfers and be able to provide us more information (like a capture of packets you can see when the BIOS is running the netboot). The Arty-35t is regularly used for this project so is know to work so I really think your issue is a network or TFTP configuration issue.

ashuthosh-mr, 2023-06-06:
Hi. That's okay. I couldn't get it to work and stopped trying. But, I worked out with serial transfer. I will have a look at Wireshark method in the future. Thanks.

closed 2024-11-20 by ashuthosh-mr