litex-hub/linux-on-litex-vexriscv #308: cc1: error: -march=rv32i2p0_ma: unsupported ISA substring '2p0_ma'

devobliquezer0, 2022-09-14:
Hi, I just set up a fresh environment with the compiler suggested in the README and I'm getting the following error log on a ./sim.py test run.

```
Build started at 2022-09-14T10:41:49.252463
Main binary: /usr/bin/python3
Build Options: -Dmultilib=false -Dpicocrt=false -Datomic-ungetc=false -Dthread-local-storage=false -Dio-long-long=true -Dformat-default=integer -Dincludedir=picolibc/riscv64-unknown-elf/include -Dlibdir=picolibc/riscv64-unknown-elf/lib '--cross-file cross.txt'
Python system: Linux
The Meson build system
Version: 0.62.2
Source dir: /litex/pythondata-software-picolibc/pythondata_software_picolibc/data
Build dir: /litex/linux-on-litex-vexriscv/build/sim/software/libc
Build type: cross build
Project name: picolibc
Project version: 1.7.8
Sanity testing C compiler: riscv64-unknown-elf-gcc
Is cross compiler: True.
Sanity check compiler command line: riscv64-unknown-elf-gcc sanitycheckc.c -o sanitycheckc_cross.exe -Os -march=rv32i2p0_ma -mabi=ilp32 -D__vexriscv__ -DUART_POLLING -g3 -fomit-frame-pointer -Wall -fno-builtin -fno-stack-protector -I/litex/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio -I/litex/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/include -I/litex/litex/litex/soc/software/libbase -I/litex/litex/litex/soc/software/include -I/litex/litex/litex/soc/software -I/litex/linux-on-litex-vexriscv/build/sim/software/include -I/litex/linux-on-litex-vexriscv/build/sim/software/include/../libc -I/litex/litex/litex/soc/cores/cpu/vexriscv_smp -fexceptions -Wpragmas -D_FILE_OFFSET_BITS=64 -c
Sanity check compile stdout:

-----
Sanity check compile stderr:
cc1: error: -march=rv32i2p0_ma: unsupported ISA substring '2p0_ma'

-----

../../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/meson.build:35:0: ERROR: Compiler riscv64-unknown-elf-gcc can not compile programs.
```

Dolu1990, 2022-09-14:
Hi,

Which version of riscv64-unknown-elf-gcc do you have ? Maybe a old one ?

280185386, 2022-10-09:
I got the same error , use this version :  riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14

280185386, 2022-10-09:
@Dolu1990  Which version should I use ?   Thank U

280185386, 2022-10-09:
gmii_ethernet] loaded (0x5626987a4130)
[xgmii_ethernet] loaded (0x5626987a4130)
[serial2tcp] loaded (0x5626987a4130)
[serial2console] loaded (0x5626987a4130)
[clocker] loaded
[ethernet] loaded (0x5626987a4130)
[spdeeprom] loaded (addr = 0x0)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Oct  9 2022 03:58:42
 BIOS CRC passed (6793857d)

 LiteX git sha1: ac369977

--=============== SoC ==================--
CPU:		VexRiscv @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> ls
Command not found
litex> cd  /
Command not found
litex> exit
Command not found
litex> cd /
Command not found
litex> 
----------------
how can I  boot a linux kernel on it ？

Dolu1990, 2022-10-11:
Hi,

I would say gcc-8.1.0 is kind of too old.
For instance me, i'm on 11.1.0.

> Booting linux

There is multiple ways, but over serial => 
https://github.com/litex-hub/linux-on-litex-vexriscv#load-the-linux-images-over-serial

enjoy-digital, 2023-01-23:
We can probably close this, the version of GCC was too old and Picolibc requires a minimal version.

closed 2023-01-23 by enjoy-digital