litex-hub/linux-on-litex-vexriscv #316: How to use rootfs from SDCard w/o Ramdisk?

pottendo, 2022-12-08:
Hi,
thanks for this project - I've managed to get Linux booting on a constrained Board (ECP5 based, https://github.com/zeldin/RVCop64, tweaked to use the VexRiscV-SMP CPU) - see attached bootlog.
I successfully built the 'buildroot' already. OpenSPI won't build (yet) here, as I'm lacking a proper cross-compiler; so I took the 'pre-built' and it worked so far.

As advertised: "32MB of RAM (Reduced to 8MB when rootfs can be put on a SDCard).", I'd like to move the rootfs to the SDCard (the board only features 16MB Ram.
I can't find the right spot, where the kernel/bootprocess is instructed to not use /dev/ram0 as rootdevice. I assume the 'bootargs' must be set set somewhere...
**Pls. let me know how to do this (properly)!
Which device name is visible under Linux?**
The SDCard is like this (could be changed if needed):
```
$ sudo fdisk /dev/sdc --list
Disk /dev/sdc: 59,48 GiB, 63864569856 bytes, 124735488 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot   Start     End Sectors  Size Id Type
/dev/sdc1          2048 2099199 2097152    1G  b W95 FAT32
/dev/sdc2       2099200 3147775 1048576  512M 83 Linux
```
- /dev/sdc1 is the boot partition, where the litex bios boots from, hosting 'Image', 'opensbi.bin', 'rv32.dtb' and 'boot.json';
- /dev/sdc2 is formatted as ext2 filesystem and hosts the unpacked rootfs.tar

boot.json:
```
$ cat /media/pottendo/F81C-7F32/boot.json 
{
	"Image":       "0x40000000",
	"rv32.dtb":    "0x40ef0000",
        "opensbi.bin": "0x40f00000"
}
```

Many thanks in advance.
pottendo

PS: Retro Commodore C64-Freaks, make sure you'll check out the 'OrangeCart' - pretty cool stuff! ;-)

[Linux-Boot.log](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10185918/Linux-Boot.log)

jeremy-heath, 2022-12-08:
Cpio extract to sdcard
Don’t load rootfs.cpio (remove from images.json)
Change dts to                 
bootargs = "mem=32M@0x40000000 rootwait console=liteuart earlycon=sbi root=/dev/mmcblk0p2 init=/sbin/init swiotlb=32";

pottendo, 2022-12-08:
Hi,Thanks for the quick response.What do you mean with "change dts to...'?Which file contains the 'bootargs = ...' line?Or is this part of "boot.json".?Sorry, I'm still new to buildroot and the RiscV / FPGA World...Bye, pottendo

bootargs = ***@***.*** rootwait console=liteuart earlycon=sbi root=/dev/mmcblk0p2 init=/sbin/init swiotlb=32";

pottendo, 2022-12-09:
hi,
thanks again, Jeremy - with your hint I recreated the dts with the right parameters:
```
[   61.375935] EXT4-fs (mmcblk0p2): re-mounted. Quota mode: disabled.
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [  120.128841] random: crng init done
OK
Starting network: OK

Welcome to Buildroot
buildroot login: root
                   __   _
                  / /  (_)__  __ ____ __
                 / /__/ / _ \/ // /\ \ /
                /____/_/_//_/\_,_//_\_\
                      / _ \/ _ \
   __   _ __      _  _\___/_//_/         ___  _
  / /  (_) /____ | |/_/__| | / /____ __ / _ \(_)__ _____  __
 / /__/ / __/ -_)>  </___/ |/ / -_) \ // , _/ (_-</ __/ |/ /
/____/_/\__/\__/_/|_|____|___/\__/_\_\/_/|_/_/___/\__/|___/
                  / __/  |/  / _ \
                 _\ \/ /|_/ / ___/
                /___/_/  /_/_/
  32-bit RISC-V Linux running on LiteX / VexRiscv-SMP with USB host support.

login[79]: root login on 'console'
root@buildroot:~# 
```
All the best,
pottendo

pottendo, 2023-02-13:
Of course this issue can be closed - br pottendo

LaneaLucy, 2024-05-29:
@pottendo would you please share what exactly you changed in the end? not only dts, everything.
thx

pottendo, 2024-05-31:
Hi @LaneaLucy, 
actually that's quite long time ago, so I can't really remember. 
Checking a bit, here some input you may investigate into:
on my boot partition (see above, 1G FAT filesystem) I have
boot.json
```
root@buildroot:/mnt# cat boot.json
{
    "Image.rv32i2p0_mafdc":       "0x40000000",
    "RVCop64-rv32i2p0_mafdc.dtb":    "0x40ef0000",
    "opensbi.bin": "0x40f00000"
}
```
(I found that there's a limit with filename-length, the the '...dtb' is close to it.)

The files there are:

```
root@buildroot:/mnt# ls -l
total 55196
-rwxr-xr-x    1 root     root       7238944 Mar 17  2023 Image.rv32i2p0_mafdc
-rwxr-xr-x    1 root     root          2621 May 31  2024 RVCop64-rv32i2p0_mafdc.dtb
-rwxr-xr-x    1 root     root           135 May 31  2024 boot.json
-rwxr-xr-x    1 root     root         53640 Dec  3  2022 opensbi.bin
```
The second partition, (root-fs) is the layout as built by buildroot.

The board's mem-layout, as configured in litex:
```
litex> mem_list
Available memory regions:
OPENSBI   0x40f00000 0x80000 
PLIC      0xf0c00000 0x400000 
CLINT     0xf0010000 0x10000 
SRAM      0x10000000 0x4000 
MAIN_RAM  0x40000000 0x1000000 
ROM       0x00000000 0xc000 
C64       0x0f000000 0x10000 
MAILBOX   0x80000000 0x40 
CSR       0xf0000000 0x10000 

```

The litex bios loads linux then like this:
```
$ /work/src/orangeCart/supporting-tools/litex/litex/litex/tools/litex_term.py /dev/ttyACM0
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: No DFU capable USB device available

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on May 31 2024 13:59:08
 BIOS CRC passed (e4ef83b6)

 LiteX git sha1: c266c379

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 80MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		48.0KiB
SRAM:		16.0KiB
MAIN-RAM:	16.0MiB

--========== Initialization ============--
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 21.8MiB/s
   Read speed: 55.4MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
Copying Image.rv32i2p0_mafdc to 0x40000000 (7238944 bytes)...
[########################################]
Copying RVCop64-rv32i2p0_mafdc.dtb to 0x40ef0000 (2621 bytes)...
[########################################]
Copying opensbi.bin to 0x40f00000 (53640 bytes)...
[########################################]
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imafdcsu
BOOT HART Features  : pmp,scounteren,mcounteren,time
BOOT HART PMP Count : 16
Firmware Base       : 0x40f00000
Firmware Size       : 124 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b109
[    0.000000] Linux version 6.1.0-rc2 (pottendo@hansi) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2022.11-rc3-30-ge87d929666) 11.3.0, GNU ld (GNU Binutils) 2.38) #7 SMP Fri Mar 17 18:18:59 CET 2023
[    0.000000] earlycon: liteuart0 at I/O port 0x0 (options '')
[    0.000000] Malformed early option 'console'
[    0.000000] earlycon: liteuart0 at MMIO 0xf0001000 (options '')
[    0.000000] printk: bootconsole [liteuart0] enabled
...
```

find attached the devicetree source, which I compiled with 
`dtc  RVCop64-usb_acm-rv32i2p0_mafdc.dts > RVCop64-usb_acm-rv32i2p0_mafdc.dtc`

bye, pottendo

[RVCop64-rv32i2p0_mafdc.zip](https://github.com/user-attachments/files/15513707/RVCop64-rv32i2p0_mafdc.zip)

LaneaLucy, 2024-05-31:
I have the problem, that my self compiled stuff looks like to big (image alone 9mb instead of 7mb) and with only 8mb ram it dont load everything. Then if i try the precompiled image and opensbi from github, nothing happens after liftoff...

pottendo, 2024-06-02:
hi,
well, find attached my build-root configs - two files:
the `.config` from the toplevel buildroot dir and the one (`DOT-config-Linux-FPUOPT`) configuring the linux kernel. Actually I don't know where to put the latter exactly, but I think you can load the settings when doing `make linux-menuconfig`.

My board (orangecart) features 16MB memory - I've configured the CPU to support both FPU and also the compact instructions (RVC) and some other processor features. But all 32bit RiscV.
I've disabled networking also to save memory, but could enable it again ending up with the image:
`Image.rv32i2p0_mafdc` size `7238944`.

Note that my last build is quite a while ago (1+ year) - As I've completely re-installed my Linux host (away from Ubuntu to ArchLinux) I've not done any build with buildroot. A simple 'make linux-menuconfig` failed already... need to debug this.

However, 8MB seems to be very low - maybe not possible with Linux unless totally reduced.
You may consider using an OS like [Zephyr](https://docs.zephyrproject.org/latest/).
I did port it to my board, see [here](https://github.com/pottendo/RVCop64-pottendo)

hope this helps, pottendo
[build-root-cfg.zip](https://github.com/user-attachments/files/15524544/build-root-cfg.zip)

pottendo, 2024-06-03:
@LaneaLucy , I managed to build again (now on my new host setup). I fell into the same openbsi issue with _zicsr_ extensions. 
As this is fixed my kernel boots even - however it grew to 8.7MB (incl. networking, without IPv6). I used gcc 14.1. 
I tried to reduce linux to the max, still got a result 1.4M larger than my previous build.
This is not good, as my test-application now crashes due to lack of memory...
...so I assume that a 8MB system cannot work with the stock linux setup as provided here.
bye, pottendo

LaneaLucy, 2024-06-03:
Then maybe someone should edit the readme...

trabucayre, 2024-06-03:
In fact buildroot provided [here](https://github.com/enjoy-digital/litex_hw_ci/tree/main/linux) must be backported to have a full env up to date and working.

LaneaLucy, 2024-06-12:
Interesting fact I found out the hard way:
If you just extract the cpio to a new partition, all file ownership gets screwed up so you actually need to use the cpio tool as root to extract