litex-hub/linux-on-litex-vexriscv #317: Where is MAC address determined?

matsbror, 2022-12-09:
Is the MAC address of the Ethernet interface determined when the FPGA bitstream is generated or is it determined in code? I have built two different versions with different MAC-addresses but I need the MAC address to be fixed because of university network rules.

jeremy-heath, 2022-12-09:
Add MAC address to the ethernet portion of the dts file then recompile.  
"local-mac-address = [00 0a 35 00 00 01];"

matsbror, 2022-12-11:
Thanks but I think I still need more info on this. As far as I know, the dts file is generated by vivado. I am using `make.py` as the only step to compile, but that would overwrite the dts file.

Then, I am still wondering what determines the mac-address as it is? Is it random?

jeremy-heath, 2022-12-11:
linux-on-litex-vexriscv generates the dts automatically and places it itn the ./build/<boardname>/<boardname>.dts
There will be an ethernet section for each ethernet component; place "local-mac-address = [00 0a 35 00 00 01];" there.

There are "locally administered mac addresses" and manufacturer administered mac addresses.  You need to pick based on your application.

matsbror, 2022-12-12:
I'm guessing this is the Ethernet section:

```
            mac0: mac@f0002000 {
                compatible = "litex,liteeth";
                reg = <0xf0002000 0x7c>,
                      <0xf0002800 0x0a>,
                      <0x80000000 0x2000>;
                reg-names = "mac", "mdio", "buffer";
                litex,rx-slots = <2>;
                litex,tx-slots = <2>;
                litex,slot-size = <2048>;
                interrupts = <2>;
                status = "okay";
            };
```

There's nothing else that it could be (I have attached the full dts [here](https://github.com/litex-hub/linux-on-litex-vexriscv/files/10204878/nexys4ddr.dts.txt)).

However, if I add a mac-address to this section, how do I then recompile to use it as I only know how to run the `make.py` command which will overwrite this file with a new .dts-file.

enjoy-digital, 2023-01-23:
@matsbror: In you case, you are using the LiteEth MAC attached to the CPU, that lets the CPU handle the MAC address by software.
I haven't tried myself @jeremy-heath's change, but to add this, you can either:
- Add it manually to the generated `.dts `and compile the `.dts `manually.
- Modify [litex_json2dts_linux.py](https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts_linux.py#L313) tool to integrate the changes. Since `make.py ` is calling this tool, this would avoid any other manual overwrite.