litex-hub/linux-on-litex-vexriscv #318: Linux in simulation

hparpinel, 2022-12-23:
How long should it takes to run Linux in simulation?
I am running the ./sim.py and it seems to be stuck at   [ 0.665874] io scheduler kyber registered
is there something I should do or just let it run?

Dolu1990, 2022-12-23:
some time, there could be ~1 seconds of simulated time between linux consol messages, so, like 15 mn of  simulation time

hparpinel, 2023-01-04:
Once I flash the board with Linux how can I check the speed of the CPU? I tried using  the command nmeter but it throws the error command not found

Dolu1990, 2023-01-05:
hmm by speed you mean frequency ? or performance ? If you need binaries, you need to enable them in buildroot and regenerate the rootfs

enjoy-digital, 2023-01-23:
@hparpinel: As pointed by @Dolu1990, with buildroot, you'll have to select/install the packages you are used to on regular distro manually.  @Dolu1990 got Debian support working for NaxRiscv (64-bit) and @gsomlo got Fedora support on Rocket (64-bit) so we should be able to provide convenient alternatives soon for developers willing to use regular distros.

closed 2023-01-23 by enjoy-digital