litex-hub/linux-on-litex-vexriscv #320: Can't open input file Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm

rybenkov, 2023-01-04:
Hi, 
thanks for this project.

I'm trying to create a bit file for an icesugar pro board using the command:

`./make.py --board=icesugar_pro --build`

but the following error occures

```console
Yosys 0.24+10 (git sha1 69cbef966, gcc 12.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fPIC -Os)

-- Executing script file `icesugar_pro.ys' --

1. Executing Verilog-2005 frontend: /home/er/Documents/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/Ram_1w_1rs_Generic.v
Parsing Verilog input from `/home/er/Documents/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/Ram_1w_1rs_Generic.v' to AST representation.
Storing AST representation for module `$abstract\Ram_1w_1rs'.
Successfully finished Verilog frontend.
ERROR: Can't open input file `/home/er/Documents/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v' for reading: No such file or directory
Traceback (most recent call last):
  File "/home/er/Documents/litex/linux-on-litex-vexriscv/./make.py", line 893, in <module>
    main()
  File "/home/er/Documents/litex/linux-on-litex-vexriscv/./make.py", line 866, in main
    builder.build(run=args.build, build_name=board_name)
  File "/home/er/Documents/litex/litex/litex/soc/integration/builder.py", line 366, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/er/Documents/litex/litex/litex/soc/integration/soc.py", line 1307, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/er/Documents/litex/litex/litex/build/lattice/platform.py", line 48, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/er/Documents/litex/litex/litex/build/lattice/trellis.py", line 67, in build
    return YosysNextPNRToolchain.build(self, platform, fragment, **kwargs)
  File "/home/er/Documents/litex/litex/litex/build/yosys_nextpnr_toolchain.py", line 126, in build
    return GenericToolchain.build(self, platform, fragment, **kwargs)
  File "/home/er/Documents/litex/litex/litex/build/generic_toolchain.py", line 118, in build
    self.run_script(script)
  File "/home/er/Documents/litex/litex/litex/build/yosys_nextpnr_toolchain.py", line 232, in run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.
```

Dolu1990, 2023-01-06:
Do you have SBT installed ?
sbt --version
 ?

rybenkov, 2023-01-08:
1) sbt --version
```code
sbt version in this project: 1.8.0
sbt script version: 1.8.0
```
2) I commented out the line `soc_kwargs = {"l2_size" : 2048}` in the file `make.py`:
```code
class IcesugarPro(Board):
    # soc_kwargs = {"l2_size" : 2048} # Use Wishbone and L2 for memory accesses.
    def __init__(self):
        from litex_boards.targets import muselab_icesugar_pro
        Board.__init__(self, muselab_icesugar_pro.BaseSoC, soc_capabilities={
            # Communication
            "serial",
            # Storage
            "spiflash",
            "sdcard",
        })
```
and got the following output to the console

```code
...
Info: Max frequency for clock '$glbnet$main_crg_clkout0': 61.76 MHz (PASS at 50.00 MHz)

Info: Max delay <async>                          -> posedge $glbnet$main_crg_clkout0: 6.99 ns
Info: Max delay posedge $glbnet$main_crg_clkout0 -> <async>                         : 5.82 ns

Info: Slack histogram:
Info:  legend: * represents 137 endpoint(s)
Info:          + represents [1,137) endpoint(s)
Info: [  3809,   7697) |*****+
Info: [  7697,  11585) |***************************+
Info: [ 11585,  15473) |***********************************+
Info: [ 15473,  19361) |************************************************************ 
Info: [ 19361,  23249) | 
Info: [ 23249,  27137) | 
Info: [ 27137,  31025) | 
Info: [ 31025,  34913) | 
Info: [ 34913,  38801) | 
Info: [ 38801,  42689) | 
Info: [ 42689,  46577) | 
Info: [ 46577,  50465) | 
Info: [ 50465,  54353) | 
Info: [ 54353,  58241) | 
Info: [ 58241,  62129) | 
Info: [ 62129,  66017) | 
Info: [ 66017,  69905) | 
Info: [ 69905,  73793) | 
Info: [ 73793,  77681) |+
Info: [ 77681,  81569) |+

Info: Program finished normally.
build/icesugar_pro/icesugar_pro.dts:44.42-48.19: Warning (interrupt_provider): /cpus/cpu@0/interrupt-controller: Missing #address-cells in interrupt provider
```

Dolu1990, 2023-01-09:
ahh nice, so sbt was missing ? + l2 cache created issues ?

rybenkov, 2023-01-16:
1) Sbt was fine.
2) I still can't build project with L2 cache due to missing 
```VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v ``` file
3) If I comment out the ```soc_kwargs``` line, then the project builds because another ```VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw16_Ood``` file is required
4) What is the difference between the files?

rybenkov, 2023-01-16:
OS: Manjaro Linux x86_64  
Kernel: 5.15.85-1-MANJARO

Regards
Eugene

Dolu1990, 2023-01-16:
So, mostly, in litex/litex/soc/cores/cpu/vexriscv_smp
There is  :

```python
cmd = 'cd {path} && sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen {args}"'.format(path=os.path.join(vdir, "ext", "VexRiscv"), args=" ".join(gen_args))
        if os.system(cmd) != 0:
            raise OSError('Failed to run sbt')
```

Which run the VexRiscv generation.

could you try with : 

```python
cmd = 'cd {path} && sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen {args}"'.format(path=os.path.join(vdir, "ext", "VexRiscv"), args=" ".join(gen_args))
        subprocess.check_call(cmd, shell=True)
```

instead ? to see how the error change

Dolu1990, 2023-01-16:
Ahh you need to make the indentation python friendly for it to work.

rybenkov, 2023-01-16:
I fixed the indentation, imported the submodule 
```python 
import subprocess
``` 
and got the following message
```code
[error] [launcher] error during sbt launcher: java.lang.UnsupportedOperationException: The Security Manager is deprecated and will be removed in a future release
Traceback (most recent call last):
  File "/home/er/Documents/test_litex/linux-on-litex-vexriscv/./make.py", line 929, in <module>
    main()
  File "/home/er/Documents/test_litex/linux-on-litex-vexriscv/./make.py", line 902, in main
    builder.build(run=args.build, build_name=board_name)
  File "/home/er/Documents/test_litex/litex/litex/soc/integration/builder.py", line 332, in build
    self.soc.finalize()
  File "/home/er/Documents/test_litex/litex/litex/soc/integration/soc.py", line 1286, in finalize
    Module.finalize(self)
  File "/home/er/Documents/test_litex/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/er/Documents/test_litex/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/er/Documents/test_litex/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/er/Documents/test_litex/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/home/er/Documents/test_litex/litex/litex/soc/cores/cpu/vexriscv_smp/core.py", line 505, in do_finalize
    self.add_sources(self.platform)
  File "/home/er/Documents/test_litex/litex/litex/soc/cores/cpu/vexriscv_smp/core.py", line 369, in add_sources
    self.generate_netlist()
  File "/home/er/Documents/test_litex/litex/litex/soc/cores/cpu/vexriscv_smp/core.py", line 275, in generate_netlist
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /home/er/Documents/test_litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv && sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=1 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=16 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=True --fpu=False --cpu-per-fpu=4 --rvc=False --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm --netlist-directory=/home/er/Documents/test_litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog --dtlb-size=4 --itlb-size=4"' returned non-zero exit status 1.

```

Dolu1990, 2023-01-16:
Nice thanks :D:D

I pushed two fixes :
- Silent generation error
- [error] [launcher] error during sbt launcher (updated SBT)

So you will need to update your local copy of : 
- https://github.com/litex-hub/pythondata-cpu-vexriscv_smp (including git submodules)
- https://github.com/enjoy-digital/litex/pull/1571

Will merge the PR as soon as it pass tests

closed 2023-01-23 by rybenkov