litex-hub/linux-on-litex-vexriscv #321: Memory initialization failed on Alveo U250

skylayer, 2023-01-09:
### Description:

I am using the litex-hub/linux-on-litex-vexriscv repository with an Alveo U250 FPGA board. I have successfully built the bitstream and loaded it onto the FPGA board, but I am having issues with initializing the SDRAM. The output of the system indicates that "Memory initialization failed".

### Steps to reproduce:

Follow the instructions in the litex-hub/linux-on-litex-vexriscv repository to build and load the bitstream onto the Alveo U250 FPGA board.

`./make.py --board=alveo_u250 --cpu-count=1 --build --load`

### Expected behavior:

The SDRAM should be successfully initialized.

### Actual behavior:

The SDRAM initialization fails, and the output of the system indicates that "Memory initialization failed".

Serial output:
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (d2ed96a8)

 LiteX git sha1: 0440733f

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64KiB
SRAM:           6KiB
SDRAM:          1048576KiB 64-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 596
  Cmd/Clk scan (0-298)
  |00001  |000001111  |000000000  |000000000| best: 180
  Setting Cmd/Clk delay to 180 taps.
  Data scan:
  m0: |00001111111111111111110| delay: 53
  m1: |01111111111111111110000| delay: 09
  m2: |00000011111111111111111| delay: 81
  m3: |11111111111111000000000| delay: 00
  m4: |11111111100000000000000| delay: -
  m5: |11111111111100000000000| delay: 00
  m6: |00000011111111111111111| delay: 81
  m7: |11111111111111000000000| delay: 00
Write latency calibration:
m0:0 m1:4 m2:4 m3:4 m4:4 m5:4 m6:4 m7:6 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b06 delays: -
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b06 delays: -
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b06 delays: -
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |00000000000000000000000000000000| delays: -
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b05 delays: -
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b06 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00000000000000000000000000000000| delays: -
  m6, b05: |00000000000000000000000000000000| delays: -
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b06 delays: -
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  139/256
  addr errors: 0/8192
  data errors: 522726/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

### System information:

FPGA board: Alveo U250 (DDR4: MICRON MTA18ASF2G72PZ-2G3B1)
Toolchain: Vivado 2020.2

### Additional context:

I have already tried the following troubleshooting steps:

Double-checking that I have followed the instructions in the repository carefully and that I have all the necessary dependencies installed.
Checking the physical connection of the SDRAM on the FPGA board.

enjoy-digital, 2023-01-23:
Hi @skylayer,

sorry, unfortunately I don't have an Alveo U250 to investigate but will try to do some tests on a similar hardware (Ultrascale + DDR4).

offnaria, 2023-04-24:
Hello @skylayer.
I also had had a problem like yours and found a solution for it.
If you still have the problem, please try my solution...

First, please add `cmd_latency = 1` option into your `litex_boards/targets/xilinx_alveo_u250.py` like [this](https://github.com/litex-hub/litex-boards/pull/500/files).

After that, build your SoC with `--with-wishbone-memory` option like:
`./make.py --board=alveo_u250 --cpu-count=1 --build --load --with-wishbone-memory`

Then, I think your SoC will pass the memtest.
If there are still errors, please try to rebooting the SoC for several times.

This is my boot log.
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (92e5b25a)

 LiteX git sha1: fb1cd22a

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64.0KiB
SRAM:           6.0KiB
L2:             2.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 584
  Cmd/Clk scan (0-292)
  |00111  |001111111  |111111111  |111111111| best: 112
  Setting Cmd/Clk delay to 112 taps.
  Data scan:
  m0: |00000111111111111111111| delay: 68
  m1: |00011111111111111111110| delay: 44
  m2: |00000011111111111111111| delay: 86
  m3: |11111111111111110000000| delay: 00
  m4: |11111111111000000000000| delay: 00
  m5: |11111111111111000000000| delay: 00
  m6: |00000001111111111111111| delay: 98
  m7: |11111111111111110000000| delay: 00
Write latency calibration:
m0:6
m1:6
m2:6
m3:6
m4:6
m5:6
m6:6
m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111100000000000000000000| delays: 88+-88
  m0, b04: |00000000000000111111111111111100| delays: 346+-129
  m0, b05: |00000000000000000000000000000000| delays: 510+-00
[... 75 lines cut ...]
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading /local/yamada/sync/projects/linux-on-litex-vexriscv/images/Image to 0x40000000 (7531468 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.9KB/s).
[LITEX-TERM] Uploading /local/yamada/sync/projects/linux-on-litex-vexriscv/images/rv32.dtb to 0x40ef0000 (1999 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.8KB/s).
[LITEX-TERM] Uploading /local/yamada/sync/projects/linux-on-litex-vexriscv/images/rootfs.cpio to 0x41000000 (3781632 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.9KB/s).
[LITEX-TERM] Uploading /local/yamada/sync/projects/linux-on-litex-vexriscv/images/opensbi.bin to 0x40f00000 (53640 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.9KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
```