litex-hub/linux-on-litex-vexriscv #322: Running linux via serial prints weird characters to output

roby2014, 2023-01-20:
I was able to compile the bitstream for the Colorlight 5a-75 board, however, when uploading linux image via serial, when I press some key I get random output (only `0` and `>`): 

[image]

Any idea what could this be? I can provide more info if needed

roby2014, 2023-01-20:
The changes I made:

```python
class Colorlight_5a_75e(Board):
    #soc_kwargs = {"l2_size" : 2048}
    def __init__(self):
        from litex_boards.targets import colorlight_5a_75x
        Board.__init__(self, colorlight_5a_75x.BaseSoC, soc_capabilities={
            # Communication
            "serial",
            #"ethernet", # had to comment because it would throw error: You cannot use the 25MHz clock as system clock since it is provided by the Ethernet PHY and will stop during PHY reset.
        })

# ....

    "colorlight_i5"               : Colorlight_i5,
    "5a_75e"               : Colorlight_5a_75e, #underscore so verilog can compile
```

```
        # CRG --------------------------------------------------------------------------------------
        with_rst     = kwargs["uart_name"] not in ["serial", "crossover"] # serial_rx shared with user_btn_n.
        with_usb_pll = kwargs.get("uart_name", None) == "usb_acm"
        self.crg = _CRG(platform, sys_clk_freq,
            use_internal_osc = use_internal_osc,
            with_usb_pll     = with_usb_pll,
            with_rst         = with_rst,
            sdram_rate       = sdram_rate
        )
``` I had to comment all these lines from `/litex_boards/targets/colorlight_5a_75x.py`, because running `./make.py --board=5a_75e --build` would throw: 
```
❯ ./make.py --board=5a_75e --build
Traceback (most recent call last):
  File "/home/roby/repos/riscv-colorlight/linux-on-litex-vexriscv/./make.py", line 940, in <module>
    main()
  File "/home/roby/repos/riscv-colorlight/linux-on-litex-vexriscv/./make.py", line 862, in main
    soc = SoCLinux(board.soc_cls, **soc_kwargs)
  File "/home/roby/repos/riscv-colorlight/linux-on-litex-vexriscv/soc_linux.py", line 160, in SoCLinux
    return _SoCLinux(**kwargs)
  File "/home/roby/repos/riscv-colorlight/linux-on-litex-vexriscv/soc_linux.py", line 34, in __init__
    soc_cls.__init__(self,
  File "/home/roby/repos/riscv-colorlight/litex-boards/litex_boards/targets/colorlight_5a_75x.py", line 138, in __init__
    with_rst     = kwargs["uart_name"] not in ["serial", "crossover"] # serial_rx shared with user_btn_n.
KeyError: 'uart_name'
```

enjoy-digital, 2023-01-23:
Hi @roby2014,

can you try to see if you are able to get the target from LiteX-Boards working?: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/colorlight_5a_75x.py

roby2014, 2023-01-24:
Hello, I was able to run `vexriscv` soft core running on the board (which uses litex_boards): https://github.com/roby2014/risc-v-colorlight-5a-75e/blob/master/base.py

So I think it works