litex-hub/linux-on-litex-vexriscv #323: Same design not working of identical boards

Beauxrel, 2023-01-26:
I am working on porting a new board (VCU108) and have 2 identical boards, for some reason the design / bitstream works on one board and allows me to load into linux. But the other board is freezing / hanging at Liftoff (Shown below). I have tried fixing timing issues and the design runs in Vivado show timing is being made, however I have suspicions about the timing wizard and asynchronous clocks. What could be the issue?

///////////////////////
Non-Working Board
///////////////////////
[image]
[image]

///////////////////////
Working Board
///////////////////////
[image]
[image]

enjoy-digital, 2023-02-13:
Hi @Beauxrel,

could you share the full LiteX BIOS prompt for the two boards? I would like to have a closer look at the DRAM calibration and compare it between boards.

enjoy-digital, 2024-07-09:
Closing since no feedback and/or hardware available.

closed 2024-07-09 by enjoy-digital

Beauxrel, 2024-07-09:
@enjoy-digital This was a board issue. I had to RMA the board!

enjoy-digital, 2024-07-09:
@Beauxrel: Ah OK, good to know, thanks for the feedback.