litex-hub/linux-on-litex-vexriscv #324: Pwmchip0 missing from directory

Beauxrel, 2023-02-10:
/sys/class/pwm is there, but I am missing pwmchip0? I am attempting to give pwm commands through pwm to control a servo.

enjoy-digital, 2023-02-13:
@Beauxrel: I would need to have a closer look but I would recommend checking the generated `.dts` to see of the PWM module is correctly defined.

Beauxrel, 2023-02-25:
/dts-v1/;

 

/ {

        #address-cells = <1>;
        #size-cells    = <1>;

        chosen {
            bootargs = "console=liteuart earlycon=liteuart,0xf0001000 rootwait root=/dev/ram0";
            linux,initrd-start = <0x41000000>;
            linux,initrd-end   = <0x41800000>;
        };
        cpus {
            #address-cells = <1>;
            #size-cells    = <0>;
            timebase-frequency = <125000000>;

            CPU0: cpu@0 {
                device_type = "cpu";
                compatible = "riscv";
                riscv,isa = "rv32i2p0_ma";
                mmu-type = "riscv,sv32";
                reg = <0>;
                clock-frequency = <125000000>;
                status = "okay";
                d-cache-size = <4096>;
                d-cache-sets = <1>;
                d-cache-block-size = <64>;
                i-cache-size = <4096>;
                i-cache-sets = <1>;
                i-cache-block-size = <64>;
                d-tlb-size = <4>;
                d-tlb-sets = <4>;
                i-tlb-size = <4>;
                i-tlb-sets = <4>;

                L0: interrupt-controller {
                    #interrupt-cells = <0x00000001>;
                    interrupt-controller;
                    compatible = "riscv,cpu-intc";
                };
            };
        };

        memory@40000000 {
            device_type = "memory";
            reg = <0x40000000 0x40000000>;
        };

        reserved-memory {
            #address-cells = <1>;
            #size-cells    = <1>;
            ranges;
            opensbi@40f00000 {
                reg = <0x40f00000 0x80000>;
            };
        };
        clocks {
            sys_clk: litex_sys_clk {
                #clock-cells = <0>;
                compatible = "fixed-clock";
                clock-frequency = <125000000>;
            };
        };

        soc {
            #address-cells = <1>;
            #size-cells    = <1>;
            compatible = "simple-bus";
            interrupt-parent = <&intc0>;
            ranges;
            soc_ctrl0: soc_controller@f0000000 {
                compatible = "litex,soc-controller";
                reg = <0xf0000000 0xc>;
                status = "okay";
            };
            lintc0: clint@f0010000 {
                compatible = "riscv,clint0";
                interrupts-extended = <&L4 3 &L4 7>;
                reg = <0xf0010000 0x10000>;
                reg-names = "control";
            };
            intc0: interrupt-controller@f0c00000 {
                compatible = "sifive,fu540-c000-plic", "sifive,plic-1.0.0";
                reg = <0xf0c00000 0x400000>;
                #address-cells = <0>;
                #interrupt-cells = <1>;
                interrupt-controller;
                interrupts-extended = <
                    &L0 11 &L0 9>;
                riscv,ndev = <32>;
            };
            liteuart0: serial@f0001000 {
                compatible = "litex,liteuart";
                reg = <0xf0001000 0x100>;
                interrupts = <0>;
                status = "okay";
            };
            leds: gpio@f0002800 {
                compatible = "litex,gpio";
                reg = <0xf0002800 0x4>;
                gpio-controller;
                #gpio-cells = <2>;
                litex,direction = "out";
                status = "disabled";
            };
            rgb_led_r0: pwm@f0004000 {
                compatible = "litex,pwm";
                reg = <0xf0004000 0x24>;
                clock = <100000000>;
                #pwm-cells = <3>;
                status = "okay";
            };
            rgb_led_g0: pwm@f0003800 {
                compatible = "litex,pwm";
                reg = <0xf0003800 0x24>;
                clock = <100000000>;
                #pwm-cells = <3>;
                status = "okay";
            };
            rgb_led_b0: pwm@f0003000 {
                compatible = "litex,pwm";
                reg = <0xf0003000 0x24>;
                clock = <100000000>;
                #pwm-cells = <3>;
                status = "okay";
            };
        };

        aliases {
                serial0 = &liteuart0;
        };
};

&leds {
        litex,ngpio = <4>;
        status = "okay";
};

Beauxrel, 2023-02-25:
@enjoy-digital, I believe the dts file is correct. Do I need to regenerate the .dtb file. If so how can I do that, I attempted to use make menuconfig and setting the in-tree/out-tree device setting (Not quite sure which on is the one I need), but Im not sure where the .dtb file is after i run make.

Beauxrel, 2023-02-25:
First Attempt at compiling my own dtb, for future reference.
```
# device tree decompile
dtc -I dtb -O dts source.dtb -o source.dts
fdtdump source.dtb > source.dts

# device tree compile
dtc -I dts -O dtb source.dts -o source.dtb_new
```

Beauxrel, 2023-03-06:
This works, for buidling new dtb files, But now I am running into an issue with the period of the linux PWM driver. Can you confirm that the PWM will provide a nice PWM signal. 

[image]