litex-hub/linux-on-litex-vexriscv #325: Adding ULX4M-LD-V2

goran-mahovlic, 2023-02-11:
Hi, 

I have made initial pull request for adding ULX4M-LD-V2 board

I have also added some changes here:

https://github.com/goran-mahovlic/linux-on-litex-vexriscv/blob/master/make.py#L417

On my old linux-on-litex everything works but I have tried to update everything and now I am getting this error

```
  File "/home/goran/Projects/litex/litex/soc/integration/builder.py", line 228, in _generate_includes
    sdram_contents = get_sdram_phy_c_header(
                     ^^^^^^^^^^^^^^^^^^^^^^^
TypeError: get_sdram_phy_c_header() takes 2 positional arguments but 3 were given
```

So if you could please check why this what else is needed to get it work...

goran-mahovlic, 2023-02-13:
> Thanks @goran-mahovlic, the LiteX-Boards support has been merged and modified a bit. The Linux-on-LiteX-Vexriscv has also been merged and I don't have the LiteX build issue you are seeing. I think LiteDRAM is probably not updated correctly on your machine.
> 
> So with this, the LiteX build seems fine, but we then have a Yosys/NextPnr issue regarding the DDR3:
> 
> ```
> ERROR: DQS group mismatch, port DQSW270 of 'ODDRX2DQA_9' in group RDQ89 is driven by DQSBUFM 'DQSBUFM_1' in group RDQ41
> ERROR: Packing design failed.
> ```
> 
> The DDR3 pin definition should probably be checked. I'm not able to do it now but could probably do it later this week.

Great, I will now pull fresh version and check everything. I think that needed fix for DDR3 never ended upstream.

goran-mahovlic, 2023-02-13:
It builds!

This was the quick fix you suggested for ULX4M

https://github.com/goran-mahovlic/litedram/commit/7440224ec8d5c7e575c2ffabfcfc16f25717dee3

And this is my mistake as I wanted to double the speed of DDR3 but it will trow error - was set to 2.0 then it builds

https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/radiona_ulx4m_ld_v2.py#L85

enjoy-digital, 2024-07-09:
I think the issue is solve and we can close this now. With LiteX-Boards support now in place, we could add support in the project by listing the board in b`oards.py`.

closed 2024-07-09 by enjoy-digital