litex-hub/linux-on-litex-vexriscv #329: build/sim/sim.dts:78.13-98.11: ERROR (phandle_references): /soc: Reference to non-existent node or label "intc0"

quantrpeter, 2023-03-04:
[image]

and meson seem has error:

[image]

thanks

quantrpeter, 2023-03-08:
[image]

enjoy-digital, 2023-03-10:
Thanks @quantrpeter, this was a regression introduced when adding Rocket support in the json2dts script, this is fixed with https://github.com/enjoy-digital/litex/commit/7bd98cf5d974799528f1b4441fd39425fd9c4c2e.

closed 2023-03-10 by enjoy-digital

quantrpeter, 2023-03-10:
thank you, it works now