litex-hub/linux-on-litex-vexriscv #332: sim.py fails for Prebuilt Image

disdi, 2023-03-12:
Downloaded #164 and extracted to images directory as shown below:

$ :/disdi/linux-on-litex-vexriscv$ ls linux_2022_03_23.zip 
linux_2022_03_23.zip
$ :/disdi/linux-on-litex-vexriscv$ ls images/
boot.json  Image  opensbi.bin  rootfs.cpio  rv32.dtb

However verilator fails simulating this image:

$ :/disdi/linux-on-litex-vexriscv$ ./sim.py 
.....................
....................
verilator -Wno-fatal -O3 --cc /lhome/saksinh/disdi/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/Ram_1w_1rs_Generic.v --cc /lhome/saksinh/disdi/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw32_Ood.v --cc /lhome/saksinh/disdi/linux-on-litex-vexriscv/build/sim/gateware/sim.v  --top-module sim --exe \
	-DPRINTF_COND=0 \
	sim_init.cpp /lhome/saksinh/disdi/litex/litex/build/sim/core/veril.cpp modules.o pads.o sim.o libdylib.o parse.o \
	--top-module sim \
	 \
	-CFLAGS "-ggdb -Wall -O3   -I/lhome/saksinh/disdi/litex/litex/build/sim/core" \
	-LDFLAGS "-lpthread -Wl,--no-as-needed -ljson-c -lz -lm -lstdc++ -Wl,--no-as-needed -ldl -levent " \
	--trace \
	 \
	 \
	--unroll-count 256 \
	--output-split 5000 \
	--output-split-cfuncs 500 \
	--output-split-ctrace 500 \
	 \
	-Wno-BLKANDNBLK \
	-Wno-WIDTH \
	-Wno-COMBDLY \
	-Wno-CASEINCOMPLETE \
	--relative-includes
%Error: /lhome/saksinh/disdi/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw32_Ood.v:16520: Unsupported or unknown PLI call: $urandom
    inputArea_target = $urandom;
                       ^~~~~~~~
%Error: /lhome/saksinh/disdi/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw32_Ood.v:16521: Unsupported or unknown PLI call: $urandom
    outputArea_hit = $urandom;
                     ^~~~~~~~
%Error: /lhome/saksinh/disdi/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw32_Ood.v:16571: Unsupported or unknown PLI call: $urandom
    buffers_0 = $urandom;
                ^~~~~~~~
%Error: /lhome/saksinh/disdi/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw32_Ood.v:16572: Unsupported or unknown PLI call: $urandom
    buffers_1 = $urandom;
                ^~~~~~~~
%Error: Exiting due to 4 error(s)
        ... See the manual and https://verilator.org for more assistance.
/lhome/saksinh/disdi/litex/litex/build/sim/core/Makefile:44: recipe for target 'sim' failed
make: *** [sim] Error 1
make: Leaving directory '/lhome/saksinh/disdi/linux-on-litex-vexriscv/build/sim/gateware'

Dolu1990, 2023-03-12:
Hi,

I think it is because of the verilator version you have.
Can you try 4.2xx ?

disdi, 2023-03-12:
Compiled verilator from source to get 4.2xx and was able to boot up the Image and get buildroot filesystem showing up.

The Ubuntu distro however give [Ubuntu verilator](https://packages.ubuntu.com/search?keywords=verilator&searchon=names&suite=kinetic)  version 4.038 by default (which does not work).

Maybe we update the README for correct version of verilator to use.

Closing the bug.

closed 2023-03-12 by disdi

Dolu1990, 2023-03-13:
Nice ^^

> Maybe we update the README for correct version of verilator to use.

It is in :)
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/README.md#-installing-verilator-only-needed-for-simulation