litex-hub/linux-on-litex-vexriscv #333: Unsupported ISA substring '2p0_ma'

kaminto-1999, 2023-03-24:
I ran into this trouble when running ./sim.py.
When I check the meson-logs.txt file, I got this message:
[image]
Is there anyone having the same problem as me? Or known how to fix it
Thank a lot !

Dolu1990, 2023-03-24:
Hi, Which version of riscv64-unknown-elf-gcc do you have ?

Dolu1990, 2023-03-24:
What do you get if you do riscv64-unknown-elf-gcc  --version ?

kaminto-1999, 2023-03-25:
> What do you get if you do riscv64-unknown-elf-gcc --version ?
It said that v8.1.0.
I got this version when running this command "wget https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14.tar.gz"
[image]

kaminto-1999, 2023-03-25:
I have updated RISC-V Toolchain from SiFive compiler and it can run through the error at this moment.
But my terminal will close after running for a while.
It always stop at this nofication
[image]

closed 2023-03-25 by kaminto-1999