litex-hub/linux-on-litex-vexriscv #335: Adding spisdcard to DE0-nano does no longer fits in Logic Elements

jeremy-heath, 2023-04-01:
Before adding total Logic Elements 10,061, after adding spisdcard total Logic Elements 36,093.  I was expecting a few hundred LE.  About 1.5 years ago, DE0-nano with serial and spisdcard would fit in the FPGA.

I need help determining what is using the extra Logic Elements.

With spisdcard
Quartus II 32-bit Version : 13.0.1 Build 232 06/12/2013 SP 1 SJ Web Edition
Revision Name : de0nano
Top-level Entity Name : de0nano
Family : Cyclone IV E
Total logic elements : 36,093
    Total combinational functions : 18,962
    Dedicated logic registers : 19,833
Total registers : 19911
Total pins : 54
Total virtual pins : 0
Total memory bits : 373,632
Embedded Multiplier 9-bit elements : 8
Total PLLs : 1

Nominal
Quartus II 32-bit Version : 13.0.1 Build 232 06/12/2013 SP 1 SJ Web Edition
Revision Name : de0nano
Top-level Entity Name : de0nano
Family : Cyclone IV E
Total logic elements : 10,061
    Total combinational functions : 7,080
    Dedicated logic registers : 5,490
Total registers : 5568
Total pins : 50
Total virtual pins : 0
Total memory bits : 292,706
Embedded Multiplier 9-bit elements : 8
Total PLLs : 1

Dolu1990, 2023-04-03:
It look like the issue is related to some memories in the spisdcard not being inferred properly as block ram, but as registers.
@enjoy-digital 
Maybe some byte mask are used to write some memory ? making things hard to infer for the synthesis tool ?

enjoy-digital, 2023-04-03:
Hi @jeremy-heath, @Dolu1990,

I'll do some tests. The `spisdcard` is not really using that much logic since it's a simple SPIMaster core, but some extra BIOS software is added to handle the SDCard, so probably using more ROM. Block are are probably all used to Quartus starts using LEs for RAM and resource usage explodes...

enjoy-digital, 2023-04-03:
@jeremy-heath: If you want to do some test before I have a look, you can play with these 2 parameters:
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L616-L617
By reducing L2 size you'll maybe free up enough BlockRAMs.

jeremy-heath, 2023-04-04:
change  "l2_size" : 512 within make.py
Barely fits. 
Total logic elements : 20,150 / 22,320 ( 90 % )
    Total combinational functions : 14,569 / 22,320 ( 65 % )
    Dedicated logic registers : 14,287 / 22,320 ( 64 % )

SPI only (not spisdcard)  within make.py
Total logic elements : 8,834 / 22,320 ( 40 % )
    Total combinational functions : 7,293 / 22,320 ( 33 % )
    Dedicated logic registers : 5,435 / 22,320 ( 24 % )

Litex-boards target "terasic_de0nano.py" is also unable to build with "add_spi_sdcard".  This is probably not a linux-on-litex-vexriscv issue.

jeremy-heath, 2023-08-03:
Rolling back Litex to 12/2022 allowed spisdcard module to fit in logic.  Still unsure what was changed.
https://github.com/enjoy-digital/litex/tree/2022.12