litex-hub/linux-on-litex-vexriscv #336: Attempted to run pre-built images without veriscv sim, application hangs on liftoff

OrkunAliOzkan, 2023-04-25:
Hello! I attempted to boot the prebuilt images without using the in-built simulator through using the following commands:

1.
```
litex_sim --with-ethernet --with-sdram --cpu-type vexriscv --no-compile-gateware --threads 8
```

2.
**Load [images ](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164) (Linux+openSBI) into repository**

3.
```
litex_sim --with-ethernet --with-sdram --cpu-type vexriscv --sdram-init boot.json --threads 8
```

However I find that the application hangs on liftoff and I do not understand why.

Dolu1990, 2023-04-25:
Hi ^^
Shouldn't it be --cpu-type vexriscv_smp ?

OrkunAliOzkan, 2023-04-25:
Hia! Wow, such a speedy response! 

Sadly replacing the `cpu type` to vexriscv_smp does not resolve the issue, and I am not sure why. It remains hanging on liftoff. 

My litex_sim file is stock, and I notice my ROM usage statistic while building is at least 20% lower than what `sim.py` displays (19% from the commands above compared to 37% from `sim.py`).

OrkunAliOzkan, 2023-04-25:
Hi! All I was missing was adding `--cpu-version linux`! I will keep you updated on whether it boots into busy-box!

Now for context, I am doing this because I am trying to build the equivalent Linux environment on a rocket cored SoC, and have been not getting much progress and I do not know why. Out of curiosity, would you mind please providing documentation here on how to create the dependencies for vexriscv which are provided in the *Linux+OpenSBI* zip file in https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164? I have documented my process for building all dependencies needed to boot Linux on a rocket cored SoC in the following issues tab: https://github.com/enjoy-digital/litex/issues/1672, however I have been unsure if I have made a mistake or not. 

Absolutely any help would be very much so appreciated!

OrkunAliOzkan, 2023-04-25:
1.
Full console log down bellow:

```
[jtagremote] loaded (0x55755b9f8ef0)
[spdeeprom] loaded (addr = 0x0)
[serial2console] loaded (0x55755b9f8ef0)
[clocker] loaded
[ethernet] loaded (0x55755b9f8ef0)
[gmii_ethernet] loaded (0x55755b9f8ef0)
[serial2tcp] loaded (0x55755b9f8ef0)
[xgmii_ethernet] loaded (0x55755b9f8ef0)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 25 2023 13:30:42
 BIOS CRC passed (5785a591)

 LiteX git sha1: 0c326f0e

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		64.0MiB 32-bit @ 1MT/s (CL-2 CWL-2)
MAIN-RAM:	64.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imas
BOOT HART Features  : time
BOOT HART PMP Count : 0
Firmware Base       : 0x40f00000
[... 55 lines cut ...]
[    0.068496] futex hash table entries: 256 (order: 2, 16384 bytes, linear)
[    0.076186] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.219383] pps_core: LinuxPPS API ver. 1 registered
[    0.219919] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.221340] PTP clock support registered
[    0.224504] FPGA manager framework
[    0.237894] clocksource: Switched to clocksource riscv_clocksource
[    0.388340] NET: Registered PF_INET protocol family
[    0.390469] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.398017] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.399338] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.400567] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.401778] TCP: Hash tables configured (established 1024 bind 1024)
[    0.403210] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.404303] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.422140] Unpacking initramfs...
[    0.457955] workingset: timestamp_bits=30 max_order=14 bucket_order=0
[    0.665114] io scheduler mq-deadline registered
[    0.665970] io scheduler kyber registered
```

2.
> Generaly, it could be because opensbi has a missconfigured UART address, or something in that kind. (as that can vary between SoC)

How do I even go about solving that problem? Is that an issue regarding the CSR file?

Dolu1990, 2023-04-25:
I think the hang is normal, it will jsut take ages to unpack the initramfs :)

>  Is that an issue regarding the CSR file?

Yes
You have to check where is the uart on the simulated SoC, and then, recompile opensbi with that address in i think, i'm not exactly sure.

OrkunAliOzkan, 2023-04-25:
> I think the hang is normal, it will jsut take ages to unpack the initramfs :)

But its been hours, surely that's far too long, how can I develop a scale of scope for what is too long and what ain't because seeing it take this long honestly fills me with sadness xD I laugh but

> You have to check where is the uart on the simulated SoC, and then, recompile opensbi with that address in i think, i'm not exactly sure.

```
...
bootargs = "console=liteuart earlycon=liteuart,0x12009000 rootwait root=/dev/ram0";
...
liteuart0: serial@12009000 {
  compatible = "litex,liteuart";
  reg = <0x12009000 0x100>;
  interrupts = <0>;
  status = "okay";
};
...
```

The following above is all related content in the csv file to the uart. 

```
make CROSS_COMPILE=riscv64-linux-gnu- PLATFORM=generic 
    FW_FDT_PATH=/...path/to/.../mydtb.dtb FW_JUMP_FDT_ADDR=0x82000000

```
The command above is to cross compile opensbi for the rocket cpu.
https://github.com/riscv-software-src/opensbi/blob/master/docs/firmware/fw.md contains all of the following are the options available for when making opensbi. 

I am unsure how to "*recompile opensbi with that address in*", would you mind please elaborating? I interpreted it as specifying `FW_JUMP_FDT_ADDR=0x12009000 `
apologies

OrkunAliOzkan, 2023-04-26:
Hi, I realised that my method of generating the device tree was incorrect and that in fact for the sim use the simulators csv generation functionality (duh!)

Dolu1990, 2023-04-28:
For opensbi, for instance here is some hardcoded addresses : 
https://github.com/litex-hub/opensbi/blob/a9ce3ada56574fcdb5befae0395f2f3d33134f81/platform/litex/vexriscv/litex.c#L29