litex-hub/linux-on-litex-vexriscv #339: Unable to build for PYNQ Z2

ratzupaltuff, 2023-05-16:
I added the Board definition to make.py:
`
`

https://github.com/litex-hub/litex-boards/issues/397 This issue looks like its possible to run vexriscv on the Pynq z2. Can someone point me in the right direction?

trabucayre, 2023-05-17:
For the mentioned issue: @enjoy-digital's answer is the solution:  `SoCCore` `clk_freq` is used to have reference frequency to compute some ratio. It's true for the uart baudrate, but since `CRG` isn't updated PLL's output remains to the default frequency resulting on a wrong baudrate.

For this issue it's an other problem: `main_ram` region is added when `LiteXSoC.add_sdram` is called or by using `--with-integrated-main-ram-size` (default None): for pynq this region isn't added.

You have to keed in mind pynq is based on a zynq (with a CPU and an FPGA). Most of peripherals (uart, ram, sdcard, ethernet) are connected to the PS (CPU) side and not available/managable from FPGA. So you have to add external USB<->uart, sdcard, etc. The most problematic part is related to the RAM: with BRAM I'm not sure you have enough of memory to be able to use Linux.

ratzupaltuff, 2023-06-07:
Thank you for the clarification, that helped a lot!

closed 2023-06-07 by ratzupaltuff