litex-hub/linux-on-litex-vexriscv #340: README recipes don't work

vuokko, 2023-05-18:
My plan was quickly build RV32 for linux, test different cache features, and study how to write a board for Arria 10 board.

Without any good knowledge of the process, I just copy pasted commands happily on a fresh Ubuntu 20.04 VM

First of all GCC needed compression feature. I believe gcc should come from Sifive which would support the default instruction set features.

Second big question was what is sbt command.
And last thing was that I got stuck java exception of missing class.

Could someone repeat this experiment, take notes and update the README ?

Dolu1990, 2023-05-19:
Hi,

Which procedure exactly did you followed ? (arguments and so one, so we can test exactly the same as you)

vuokko, 2023-05-19:
I followed the README.md by starting with Prerequisites
Then I continued to Installing LiteX and Installing a RISC-V toolchain

I added those bin directories to my $PATH

Then I tried to build something with various tries
./make.py --board=arty --toolchain=symbiflow --build
./make.py --board=arty --build
./make.py --board=arty 

and other boards just to prove that I have some kind of working system.

I also needed the --with-rvc or something like that to enable compression because otherwise gcc wouldn't work.

Most of the time I was just trying to figure out what I'm missing and trying stupid things for example to get the environment to build the missing java classes. ClassNotFoundException in sbt runMain stage.

TeOSobrino, 2024-04-25:
Hi, just wanted to add on this matter, sbt is not listed in the dependencies, it's quite an obvious fix as you will be prompted with some error message alike "sbt not found in: /bin/sbt", but I have found this project to be so tidy, that adding this at the dependencies would be a nice minor improvement, as it is prossible to simulate a litex SoC without it.

Dolu1990, 2024-04-25:
Added sbt stuff to the readme ^^
Thanks