litex-hub/linux-on-litex-vexriscv #343: When I build project with "--with-ethernet" I got "Liftoff!" screen

fatihakalan, 2023-07-19:
Hey, I want to build litex project with ethernet peripheral. I succesfully builded project and generated bit file. I'm trying to loading boot.json file but  I got this error.
[image]

How do I solve this any idea ? I try the changing --sys-clk 50MHz, 75 MHz ,100MHz but didn't worked. Also I was compiled with 100 MHz memtest is failed. 

building command: python3 -m litex_boards.targets.efinix_trion_t120_bga576_dev_kit --with-ethernet --eth-ip ETH_IP --eth-phy=1  --sys-clk-freq 50e6 --build --load

Beauxrel, 2023-07-24:
Have you gotten the serial boot to work?

fatihakalan, 2023-07-24:
Yeah, I'm using serial boot way this command (litex_term --images=images/boot.json /dev/ttyUSBX) and I got BIOS menu. I run the reboot command but then stuck Liftoff screen

Beauxrel, 2023-07-24:
I see these clock values 

_io = [
    # Clk
    ("clk40", 0, Pins("P19"), IOStandard("3.3_V_LVTTL_/_LVCMOS")),
    ("clk50", 0, Pins("AA8"), IOStandard("3.3_V_LVTTL_/_LVCMOS")),
    ("clk20", 0, Pins("AA9"), IOStandard("3.3_V_LVTTL_/_LVCMOS")),
    ("clk74_25", 0, Pins("J9"), IOStandard("3.3_V_LVTTL_/_LVCMOS")),

have you tried clock 40, 20 or 74.25?

Beauxrel, 2023-07-24:
Please let me know if that worked for you. @fatihakalan

fatihakalan, 2023-07-25:
Thanks for reply first of all but I try now didn't work. 40, 20 or 74.25 MHz clocks. and also when ı give to 74.25 I get PLL error. In the other clocks I succesfully loading ".bit" file but linux doesn't boot succesfully. What should I do ? @Beauxrel

Beauxrel, 2023-07-26:
from the linux-on-litex folder @fatihakalan

Beauxrel, 2023-07-26:
Some of those commands may not be acceptable, but building the SoC from the make.py file is how I always do it.

fatihakalan, 2023-07-26:
I get this error when I run "./make.py --board=efinix_trion_t120_bga576_dev_kit --with-ethernet --eth-ip ETH_IP --eth-phy=1 --sys-clk-freq 50e6 --build --load" 
- make.py: error: unrecognized arguments: --with-ethernet --eth-ip ETH_IP --eth-phy=1 --sys-clk-freq 50e6
- @Beauxrel

fatihakalan, 2023-07-26:
When I want to add some peripheral to my SoC I follow this repo:https://github.com/litex-hub/litex-boards. make.py command ı guess doesn't work with peripheral did you ever add peripheral parameter ?
@Beauxrel

Beauxrel, 2023-07-26:
Can you copy and paste the full make.py file

fatihakalan, 2023-07-26:
#!/usr/bin/env python3

#
# This file is part of Linux-on-LiteX-VexRiscv
#
# Copyright (c) 2019-2022, Linux-on-LiteX-VexRiscv Developers
# SPDX-License-Identifier: BSD-2-Clause

import os
import sys
import argparse

from litex.soc.integration.builder import Builder
from litex.soc.cores.cpu.vexriscv_smp import VexRiscvSMP

from soc_linux import SoCLinux

# Board Definition ---------------------------------------------------------------------------------

class Board:
    soc_kwargs = {
        "integrated_rom_size"  : 0x10000,
        "integrated_sram_size" : 0x1800,
        "l2_size"              : 0
    }
    def __init__(self, soc_cls=None, soc_capabilities={}, soc_constants={}):
        self.soc_cls          = soc_cls
        self.soc_capabilities = soc_capabilities
        self.soc_constants    = soc_constants

    def load(self, filename):
        prog = self.platform.create_programmer()
        prog.load_bitstream(filename)

    def flash(self, filename):
        prog = self.platform.create_programmer()
        prog.flash(0, filename)

#---------------------------------------------------------------------------------------------------
# Xilinx Boards
#---------------------------------------------------------------------------------------------------

# Acorn support ------------------------------------------------------------------------------------

class Acorn(Board):
    soc_kwargs = {"uart_name": "jtag_uart", "sys_clk_freq": int(150e6)}
    def __init__(self):
        from litex_boards.targets import sqrl_acorn
        Board.__init__(self, sqrl_acorn.BaseSoC, soc_capabilities={
            # Communication
            "serial",
            # Storage
            "sata",
        })

# Acorn PCIe support -------------------------------------------------------------------------------

class AcornPCIe(Board):
    soc_kwargs = {"uart_name": "crossover", "sys_clk_freq": int(125e6)}
    def __init__(self):
        from litex_boards.targets import sqrl_acorn
        Board.__init__(self, sqrl_acorn.BaseSoC, soc_capabilities={
            # Communication
            "serial",
            "pcie",
        })

    def flash(self, filename):
        prog = self.platform.create_programmer()
        prog.flash(0, filename.replace(".bin", "_fallback.bin"))
[... 849 lines cut ...]
        soc.generate_dts(board_name)
        soc.compile_dts(board_name, args.fdtoverlays)

        # DTB --------------------------------------------------------------------------------------
        soc.combine_dtb(board_name, args.fdtoverlays)

        # PCIe Driver ------------------------------------------------------------------------------
        if "pcie" in board.soc_capabilities:
            from litepcie.software import generate_litepcie_software
            generate_litepcie_software(soc, os.path.join(builder.output_dir, "driver"))

        # Load FPGA bitstream ----------------------------------------------------------------------
        if args.load:
            board.load(filename=builder.get_bitstream_filename(mode="sram"))

        # Flash bitstream/images (to SPI Flash) ----------------------------------------------------
        if args.flash:
            board.flash(filename=builder.get_bitstream_filename(mode="flash"))

        # Generate SoC documentation ---------------------------------------------------------------
        if args.doc:
            soc.generate_doc(board_name)

if __name__ == "__main__":
    main()

fatihakalan, 2023-07-26:
Efinix Boards
#---------------------------------------------------------------------------------------------------

class TrionT120BGA576DevKit(Board):
soc_kwargs = {"l2_size" : 2048} # Use Wishbone and L2 for memory accesses.
def init(self):
from litex_boards.targets import efinix_trion_t120_bga576_dev_kit
Board.init(self, efinix_trion_t120_bga576_dev_kit.BaseSoC, soc_capabilities={
# Communication
"serial",
"ethernet",
# GPIOs
"leds",
})
I was add "ethernet" option at now trying with this maybe its effected . Normally there no ethernet in communication line just serial

Beauxrel, 2023-07-26:
That is excatly what I was going to say, try to include ethernet and build it without all the other commands  ./make.p board=<your board> --sys-clk-freq 50e6 --build --load

fatihakalan, 2023-07-26:
Okay I'm trying to directly build but "--load" doesn't work. I'm using OpenFPGALoader to boot .bit file. Do you think any effect ? And  I let you know my output.

Beauxrel, 2023-07-26:
Is the bit file being generated?

fatihakalan, 2023-07-26:
Yeah it is being created I'm waitig to finish

fatihakalan, 2023-07-26:
it is succesfully booting .bit file and also linux but there are no ethernet as like eth0 when I run ifconfig command.

Beauxrel, 2023-07-26:
In your efinix_trion_t120_bga576_dev_kit.py file try setting
with_ethernet, with_etherbone and eth_phy 

class BaseSoC(SoCCore):
    def __init__(self, sys_clk_freq=75e6,
        with_spi_flash  = False,
        with_ethernet   = False, <- true
        with_etherbone  = False,<- true
        eth_phy         = 0, <- 1
        eth_ip          = "192.168.1.50",
        with_led_chaser = True,
        **kwargs):
        platform = efinix_trion_t120_bga576_dev_kit.Platform()

Beauxrel, 2023-07-26:
and running make.py again

fatihakalan, 2023-07-26:
yeah base frequency 75 MHz .I'm trying now actually ı remember ı was tried this way but didn't work. I let you know output

fatihakalan, 2023-07-26:
okay I'm trying all of true and if it's fail ım gonna try case one of them is true (withethernet). But where is the if statement  you mentioned about.

Beauxrel, 2023-07-26:
`# Ethernet / Etherbone ---------------------------------------------------------------------
        if with_ethernet or with_etherbone:
            self.ethphy = LiteEthPHYRGMII(
                platform           = platform,
                clock_pads         = platform.request("eth_clocks", eth_phy),
                pads               = platform.request("eth", eth_phy),
                with_hw_init_reset = False)
            if with_ethernet:
                self.add_ethernet(phy=self.ethphy, software_debug=False)
            if with_etherbone:
                self.add_etherbone(phy=self.ethphy)

            # FIXME: Avoid this.
            platform.toolchain.excluded_ios.append(platform.lookup_request("eth_clocks").tx)
            platform.toolchain.excluded_ios.append(platform.lookup_request("eth_clocks").rx)
            platform.toolchain.excluded_ios.append(platform.lookup_request("eth").tx_data)
            platform.toolchain.excluded_ios.append(platform.lookup_request("eth").rx_data)
            platform.toolchain.excluded_ios.append(platform.lookup_request("eth").mdio)`

fatihakalan, 2023-07-26:
[image]
When I changed 

with_ethernet = False, <- true
with_etherbone = False,<- true
eth_phy = 0, <- 1

there are no eth0 @Beauxrel

fatihakalan, 2023-07-26:
I tried this 
with_ethernet =  true
with_etherbone = False
eth_phy = 0, 

but doesn't work. There are no eth0 @Beauxrel

Beauxrel, 2023-07-27:
Looks like there was a push 3 hours ago for your board using PMOD RGMII

https://github.com/litex-hub/litex-boards/commit/2d3b81a532d02d316b529139727957d5c93af737