litex-hub/linux-on-litex-vexriscv #352: How can I send files to linux?

nohahanon, 2023-09-13:
Hi there!

I wanna execute c programs in this linux environment but I have no idea that how to send files to this linux. If you knew it, please teach me!

environment: windows11+wsl2, tang-nano-20k
images: I'm using the dts and dtb files generated when building './make.py --board=sipeed_tang_nano_20k' to replace the dts and dtb files provided in 'TangNano-20k-example/linux'

Thank you.

jeremy-heath, 2023-09-13:
With an SD card mount the card and copy executable to it.  

If you have a network and NFS enabled, mount the remote directory within the local image once loaded.

With Buildroot there is a "System Configuration" option to add a pre-existing directory to the rootfs.cpio image.
 ($(BR2_EXTERNAL_LITEX_VEXRISCV_PATH)/board/litex_vexriscv/rootfs_overlay) Root system overlay directories

nohahanon, 2023-09-14:
Thank you for replying!!

Probably, I should specify the option you mentioned in the following operations: 
https://github.com/litex-hub/linux-on-litex-vexriscv#-generating-the-linux-binaries-optional
However I do not know how to add that option "System Configuration" in the command lines. 
Can you give few more advice on this ?

nohahanon, 2023-09-14:
I've captured the results of running 'dmesg' and 'ls /dev'. I've successfully managed to boot Linux using an SD card, but I'm unable to access the SD card that is inserted in the tang-nano-20k.
```
/ # dmesg
[    0.000000] Linux version 6.4.0-rc1+ (zjs@CL-A-00022) (riscv64-linux-gnu-gcc (Debian 12.2.0-13) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #10 Tue May 16 09:20:49 CST 2023
[    0.000000] earlycon: liteuart0 at I/O port 0x0 (options '')
[    0.000000] Malformed early option 'console'
[    0.000000] earlycon: liteuart0 at MMIO 0xf0001000 (options '')
[    0.000000] printk: bootconsole [liteuart0] enabled
[    0.000000] INITRD: 0x41000000+0x00800000 is not a memory region - disabling initrd
[    0.000000] OF: reserved mem: 0x40f00000..0x40f7ffff (512 KiB) map non-reusable opensbi@40f00000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x00000000407fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000407fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000407fffff]
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x1 Version=0x8
[    0.000000] SBI TIME extension detected
[    0.000000] SBI IPI extension detected
[    0.000000] SBI RFENCE extension detected
[    0.000000] riscv: base ISA extensions aim
[    0.000000] riscv: ELF capabilities aim
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Kernel command line: console=liteuart earlycon=liteuart,0xf0001000 rootwait root=/dev/ram0
[    0.000000] Dentry cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000000] Inode-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping off.  Total pages: 2032
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 6240K/8192K available (1084K kernel code, 82K rwdata, 110K rodata, 495K init, 49K bss, 1952K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-1, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] riscv-intc: 32 local interrupts mapped
[    0.000000] plic: interrupt-controller@f0c00000: mapped 32 interrupts with 1 handlers for 2 contexts.
[    0.000000] riscv-timer: riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000082] sched_clock: 64 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.013713] Console: colour dummy device 80x25
[    0.020394] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.032445] pid_max: default: 4096 minimum: 301
[    0.039536] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.048789] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.119071] ASID allocator using 9 bits (512 entries)
[    0.152672] devtmpfs: initialized
[    0.278816] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.290399] futex hash table entries: 16 (order: -5, 192 bytes, linear)
[    0.361854] platform soc: Fixed dependency cycle(s) with /soc/interrupt-controller@f0c00000
[    0.484054] clocksource: Switched to clocksource riscv_clocksource
[    1.469552] workingset: timestamp_bits=30 max_order=11 bucket_order=0
[    2.606407] LiteX SoC Controller driver initialized
[    3.062200] f0001000.serial: ttyLXU0 at MMIO 0x0 (irq = 0, base_baud = 0) is a liteuart
[    3.078516] printk: console [liteuart0] enabled
[    3.088632] printk: bootconsole [liteuart0] disabled
[    3.301316] clk: Disabling unused clocks
[    3.307729] Warning: unable to open an initial console.
[    3.352344] Freeing unused kernel image (initmem) memory: 492K
[    3.359256] Kernel memory protection not selected by kernel config.
[    3.366560] Run /init as init process
[    3.369976]   with arguments:
[    3.370581]     /init
[    3.371303]   with environment:
[    3.371911]     HOME=/
[    3.372660]     TERM=linux
/ #
/ # ls /dev
console  tty1     tty2     tty3     tty4     tty5     tty6     vcs1
full     tty10    tty20    tty30    tty40    tty50    tty60    vcsa
kmsg     tty11    tty21    tty31    tty41    tty51    tty61    vcsa1
mem      tty12    tty22    tty32    tty42    tty52    tty62    vcsu
null     tty13    tty23    tty33    tty43    tty53    tty63    vcsu1
ptmx     tty14    tty24    tty34    tty44    tty54    tty7     zero
pts      tty15    tty25    tty35    tty45    tty55    tty8
random   tty16    tty26    tty36    tty46    tty56    tty9
shm      tty17    tty27    tty37    tty47    tty57    ttyLXU0
tty      tty18    tty28    tty38    tty48    tty58    urandom
tty0     tty19    tty29    tty39    tty49    tty59    vcs
```

jeremy-heath, 2023-09-17:
For future reference.
Add the directory  and file to $(BR2_EXTERNAL_LITEX_VEXRISCV_PATH)/board/litex_vexriscv/rootfs_overlay
for example /home/<userr>/linux-on-litex-vexriscv/board/litex_vexriscv/rootfs_overlay/opt/filename.txt
Then rebuild buildroot and the rootfs.cpio will have the  /opt/filename.txt

nohahanon, 2023-09-17:
Thank you for your response!
I will try it later. I was able to create fw_jump.bin (apparently corresponding to opensbi.bin), rootfs.cpio, and Image using Buildroot. With the device tree (dtb) file created using make.py, I attempted to boot Linux following the boot.json provided [here](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164), but I encountered an error: "[LITEX-TERM] Got unexpected response from device 'b'E''", and I couldn't reach liftoff...

I have created an [issue](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/354) for this matter. I will try the suggestions you provided once that is resolved. If you could provide advice on the linked issue, please help!

closed 2023-12-15 by nohahanon