litex-hub/linux-on-litex-vexriscv #353: Some issues when I boot on tang-primer-20k

sususjysjy, 2023-09-16:
Hi , i met some issues:
If i use this  project's codes , i can successfully build the bitstream.( include spi-sdcard and serial ) use the following command : `./make.py --board=tang_primer_20k`
but if I use Litex , i use the command `python3 sipeed_tang_primer_20k.py --cpu-type vexriscv_smp --with-spi-sdcard --build`
,the resources will not be enough.
[image]
i ' m so confused of it . if you change cores or ips in this project?

BobMa0420, 2023-10-19:
Hi, I uesd `./make.py --board=sipeed_tang_primer_20k --toolchain=Gowin --with-video-terminal --build` and got

> make.py: error: unrecognized arguments: --with-video-terminal

Do you have any idea of that? Thank you so much!

enjoy-digital, 2024-07-09:
Hi @kevinsu20,

in the project, we are reducing the L2 size to avoid resources issue on this board, so you also had to reduce L2 size in your command line: 
`python3 -m litex_boards.targets.sipeed_tang_primer_20k --cpu-type vexriscv_smp --with-spi-sdcard --l2-size=512 --build `

@BobMa0420 : I don't think the FPGA will be large enough to enable both Linux and Video Terminal.

closed 2024-07-09 by enjoy-digital