litex-hub/linux-on-litex-vexriscv #355: how to run coremark to test my cpu by litex?

1487021, 2023-10-04:
now,i have load the linux image to my board,how to run coremark to test my cpu by litex?

Dolu1990, 2023-10-09:
Following https://github.com/litex-hub/linux-on-litex-vexriscv#-generating-the-linux-binaries-optional
You can add the coremark with a make xconfig

```shell
git clone http://github.com/buildroot/buildroot
cd buildroot
make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
make xconfig # Add coremark in it
make
```

1487021, 2023-10-09:
thanks for your reply,but i have

1487021, 2023-10-09:
thanks for your reply@Dolu1990 ,but i have no idea

BobMa0420, 2023-10-09:
> Following https://github.com/litex-hub/linux-on-litex-vexriscv#-generating-the-linux-binaries-optional You can add the coremark with a make xconfig
>
> ```shell
> git clone [...]

Is `litex_vexriscv_defconfig` the .config file generrated by `make menuconfig`?

trabucayre, 2023-10-09:
Yes.
This issue (and answer) is similar to #358

trabucayre, 2023-10-09:
`.config` is the full configuration, xxx_defconfig is the bare minimal configuration set to obtain each time the same .config

BobMa0420, 2023-10-10:
> Following https://github.com/litex-hub/linux-on-litex-vexriscv#-generating-the-linux-binaries-optional You can add the coremark with a make xconfig
>
> ```shell
> git clone [...]

Hi, I use  `make BR2_EXTERNAL=/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/ 1.config`, which 1.config is generated by `make menuconfig` (select coremark package).

> Target packages  --->
    Debugging, profiling and benchmark  --->
        [*] coremark

Then I got this error:

> bobma@bobma-virtual-machine:~/python-litex/linux-on-litex-vexriscv/buildroot$ make BR2_EXTERNAL=/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/ 1.config
Makefile:192: *** '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/': does not have an 'external.desc'. See https://buildroot.org/manual.html#br2-external-converting.  Stop.
make: *** [Makefile:82: _all] Error 2

1487021, 2023-10-10:
thanks for your reply!@trabucayre now,i followed the steps to make,add the coremrk in it,then creat a 'output',what should i do next?can you give me more details,thanks for a lot!

G1oRia1, 2023-10-10:
> Following https://github.com/litex-hub/linux-on-litex-vexriscv#-generating-the-linux-binaries-optional You can add the coremark with a make xconfig
>
> ```shell
> git clone [...]

 @trabucayre @Dolu1990 Thank you for your guidance on the previous steps,now I am experiencing some problems again, can you give me some advice？

I was able to make,and after several mins of compiles,then i got a "rootfs.tar" in output/images. should i change "rootfs.cpio" and rewrite "boot.json" which i got in  [linux_2022_03_23.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/8331338/linux_2022_03_23.zip) ？
Actually,i already did that,i replaced "rootfs.cpio" with "rootfs.tar" and changed "boot.json",then i load the changed linux image to SDcard.However,it stuck at 

>[    9.045887] io scheduler mq-deadline registered
[    9.057840] io scheduler kyber registered
[    9.198371] LiteX SoC Controller driver initialized
[   20.657553] f0001000.serial: ttyLXU0 at MMIO 0x0 (irq = 0, base_baud = 0) is a liteuart
[   20.689628] printk: console [liteuart0] enabled
[   20.689628] printk: console [liteuart0] enabled
[   20.708240] printk: bootconsole [liteuart0] disabled
[   20.708240] printk: bootconsole [liteuart0] disabled
[   21.104874] i2c_dev: i2c /dev entries driver
[   21.260597] mmc_spi spi0.0: SD/MMC host mmc0, no WP, no poweroff, cd polling
[   21.660629] NET: Registered PF_INET6 protocol family
[   21.973867] Segment Routing with IPv6
[   21.988516] In-situ OAM (IOAM) with IPv6
[   22.006297] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   22.157222] NET: Registered PF_PACKET protocol family
[   22.392531] mmc0: host does not support reading read-only switch, assuming write-enable
[   22.413401] mmc0: new SDHC card on SPI
[   22.545582] mmcblk0: mmc0:0000 SD32G 29.7 GiB 
[   22.984983]  mmcblk0: p1
[   23.073748] Waiting for root device /dev/ram0...

Was what I did earlier right?If no,please tell me what should i do after i got "rootfs.tar". If yes, please let me know what my current problem is.I look forward to your responses.thanks.

BobMa0420, 2023-10-11:
> Following https://github.com/litex-hub/linux-on-litex-vexriscv#-generating-the-linux-binaries-optional You can add the coremark with a make xconfig
>
> ```shell
> git clone [...]

Hi, I run `make BR2_EXTERNAL=/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig` and I got this:

>/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-split-html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-split-html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.chunked'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.chunked'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-pdf'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-pdf'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.pdf'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.pdf'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-text'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-text'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.text'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.text'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-epub'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-epub'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.epub'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.epub'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-clean'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-clean'
Config.in:1: can't open file "/package/dhrystone-opt/Config.in"
make[1]: *** [Makefile:1013: litex_vexriscv_defconfig] Error 1
make: *** [Makefile:82: _all] Error 2

Dolu1990, 2023-10-11:
> /home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies'

Ahh no idea what it is.

> Actually,i already did that,i replaced "rootfs.cpio" with "rootfs.tar" 

I would avoid changes like this, try to keep things as much as they were before i would say.

trabucayre, 2023-10-11:
Most important part is:
```
Config.in:1: can't open file "/package/dhrystone-opt/Config.in"
make[1]: *** [Makefile:1013: litex_vexriscv_defconfig] Error 1
make: *** [Makefile:82: _all] Error 2
```
Seems `BR2_EXTERNAL_LITEX_VEXRISCV_PATH` is lost...
Could you try:
```
BR2_EXTERNAL=$(pwd)/../linux-on-litex-vexriscv/buildroot/ make litex_vexriscv_defconfig
```

rootfs.tar is an archive containing rootfs for target system. rootfs.cpio is a image to load.

BobMa0420, 2023-10-12:
> /home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/

Thanks for your reply. It seems that got a same error. Could it be a problem of the file permissions of  "/package/dhrystone-opt/Config.in"?

> BR2_EXTERNAL=/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/ make litex_vexriscv_defconfig
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-split-html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-split-html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.chunked'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.chunked'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-pdf'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-pdf'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.pdf'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.pdf'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-text'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-text'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.text'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.text'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-epub'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-epub'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.epub'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.epub'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-clean'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-clean'
Config.in:1: can't open file "/package/dhrystone-opt/Config.in"
make[1]: *** [Makefile:1013: litex_vexriscv_defconfig] Error 1
make: *** [Makefile:82: _all] Error 2

trabucayre, 2023-10-12:
If you refers to `/package/dhrystone-opt/Config.in`you talk about someting under your root (**/**) directory,
but this directory is in fact in *linux-on-litex-vexriscv/buildroot/package* and buildroot access to this one using *$(BR2_EXTERNAL_LITEX_VEXRISCV_PATH)/package/dhrystone-opt/Config.in*.

So if **BR2_EXTERNAL_LITEX_VEXRISCV_PATH** isn't defined the result is to search something immediately under */* directory.

Could you share your distribution, make version and which shell you use (bash, dash, ksh, ...) and the `.config` file located at buildroot's root directory?

Another test is to always pass `BR2_EXTERNAL=$(pwd)/../linux-on-litex-vexriscv/buildroot/` each time you use make (keep in mind to adapt path to your specific case (here buildroot directory is located at the same level as *linux-on-vexriscv*)).

BobMa0420, 2023-10-12:
> If you refers to `/package/dhrystone-opt/Config.in`you talk about someting under your root (**/**) directory, but this directory is in fact in _linux-on-litex-vexriscv/buildroot/package_ and [...]

Distribution：
> Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal

Make version：

> GNU Make 4.2.1
Built for x86_64-pc-linux-gnu

Shell:

> /bin/bash

`Config.in` located at buildroot's root directory:

> source "$BR2_EXTERNAL_LITEX_VEXRISCV_PATH/package/dhrystone-opt/Config.in"

config BR2_PACKAGE_VEXRISCV_AES
	bool "VexRiscv AES custom instruction"
	help
	  Enable VexRiscv AES custom instruction usage in openssl and dropbear

BobMa0420, 2023-10-12:
I replaced **$BR2_EXTERNAL_LITEX_VEXRISCV_PATH** in `Config.in` and `litex_vexriscv_defconfig` with **/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot** and run `make BR2_EXTERNAL=/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig`

And got

> /home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-split-html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-split-html'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.chunked'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.chunked'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-pdf'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-pdf'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.pdf'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.pdf'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-text'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-text'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.text'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.text'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-epub'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-epub'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.epub'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.epub'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-clean'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-clean'
configuration written to /home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/.config

Then run `make xconfig`, I can see two options called dhrystone-opt and VexRiscv AES custom instruction, then I enable both and save.

Next I run make

> /home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/build/docs/manual/.stamp_doc_rsynced'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-check-dependencies-html'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-check-dependencies-html'
[... cut ...] '
docs/manual/manual.mk:10: warning: ignoring old recipe for target '/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/docs/manual/manual.epub'
/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/docs/manual/manual.mk:10: warning: overriding recipe for target 'manual-clean'
docs/manual/manual.mk:10: warning: ignoring old recipe for target 'manual-clean'
Makefile:944: *** Please configure Buildroot first (e.g. "make menuconfig").  Stop.
make: *** [Makefile:82: _all] Error 2

trabucayre, 2023-10-13:
I'm dumb!
Seems you have *buildroot* installed into *linux-on-litex-vexriscv/buildroot* directory with a potential conflict / file's override.

*linux-on-litex-vexriscv/buildroot* is a **BR2_EXTERNAL** ie a plugin to extend *buildroot* by adding configs, and/or packages. Both share a similar structure and if you copy/move files from latter into former some files will be overrided.

Could you restart with a fresh *linux-on-litex-vexriscv* and a *buildroot* into two different locations?
ie:

```
.
├── linux-on-litex-vexriscv
└── buildroot
```

BobMa0420, 2023-10-13:
Thank you so much!! Seems to have a great progress.

I run `git clone http://github.com/buildroot/buildroot` and `git clone https://github.com/litex-hub/linux-on-litex-vexriscv.git`

and got 

> bobma@bobma-virtual-machine:~/temp_file$ tree -L 1
> .
> ├── buildroot
> └── linux-on-litex-vexriscv

then `cd buildroot/`

then `make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig` and got

> mkdir -p /home/bobma/temp_file/buildroot/output/build/buildroot-config/lxdialog
> PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
>     obj=/home/bobma/temp_file/buildroot/output/build/buildroot-config -C support/kconfig -f Makefile.br conf
> /usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/bobma/temp_file/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /home/bobma/temp_file/buildroot/output/build/buildroot-config/.depend 2>/dev/null || :
> /usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/bobma/temp_file/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /home/bobma/temp_file/buildroot/output/build/buildroot-config/conf.o
> /usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/bobma/temp_file/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -I. -c /home/bobma/temp_file/buildroot/output/build/buildroot-config/zconf.tab.c -o /home/bobma/temp_file/buildroot/output/build/buildroot-config/zconf.tab.o
> /usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/bobma/temp_file/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   /home/bobma/temp_file/buildroot/output/build/buildroot-config/conf.o /home/bobma/temp_file/buildroot/output/build/buildroot-config/zconf.tab.o  -o /home/bobma/temp_file/buildroot/output/build/buildroot-config/conf
> rm /home/bobma/temp_file/buildroot/output/build/buildroot-config/zconf.tab.c
> 
> configuration written to /home/bobma/temp_file/buildroot/.config

then `make xconfig`, a window open then I enable coremark option, save and close the buildroot configuration window

then `make`

got 

> package/Makefile.in:61: *** No C library enabled, this is not possible..  Stop.
> make: *** [Makefile:82: _all] Error 2

What should I do next? Thank you sooo much again!!

trabucayre, 2023-10-13:
It's weird: I have fixed this issue with [this commit](https://github.com/litex-hub/linux-on-litex-vexriscv/commit/108c020e5d0746e9d53a70931709516cfe4367e8)
Could you check:
```
Target options  --->
    -*- Base Integer (I)
    [*] Integer Multiplication and Division (M)
    [*] Atomic Instructions (A)
```
And if not checked -> check

BobMa0420, 2023-10-13:
I run `make menuconfig` and enable **Integer Multiplication and Division (M) & Atomic Instructions (A)**

then `make` and I am waitting for it to finish running now.

BobMa0420, 2023-10-13:
I got these files after running `make`

> bobma@bobma-virtual-machine:~/temp_file/linux-on-litex-vexriscv/images$ ls -all
> total 22932
> drwxrwxr-x 2 bobma bobma     4096 10月 13 18:57 .
> drwxrwxr-x 9 bobma bobma     4096 10月 12 10:21 ..
> -rw-rw-r-- 1 bobma bobma      123 10月 11 11:19 boot.json
> -rw-r--r-- 1 bobma bobma 52428800 10月 13 18:47 boot.vfat
> lrwxrwxrwx 1 bobma bobma       51 10月 13 18:47 Image -> /home/bobma/temp_file/buildroot/output/images/Image
> lrwxrwxrwx 1 bobma bobma       57 10月 13 18:47 opensbi.bin -> /home/bobma/temp_file/buildroot/output/images/fw_jump.bin
> lrwxrwxrwx 1 bobma bobma       57 10月 13 18:47 rootfs.cpio -> /home/bobma/temp_file/buildroot/output/images/rootfs.cpio
> -rw-rw-r-- 1 bobma bobma     2613 10月 12 10:24 rv32.dtb
> -rw-r--r-- 1 bobma bobma 52429312 10月 13 18:47 sdcard.img

and

> bobma@bobma-virtual-machine:~/temp_file/buildroot/output/images$ ls -all
> total 16064
> drwxr-xr-x 2 bobma bobma    4096 10月 13 18:47 .
> drwxrwxr-x 6 bobma bobma    4096 10月 13 18:47 ..
> -rw-r--r-- 1 bobma bobma  263644 10月 13 18:46 fw_jump.bin
> -rw-r--r-- 1 bobma bobma  820876 10月 13 18:46 fw_jump.elf
> -rw-r--r-- 1 bobma bobma 7742944 10月 13 18:47 Image
> -rw-r--r-- 1 bobma bobma 3592704 10月 13 18:47 rootfs.cpio
> -rw-r--r-- 1 bobma bobma 4136960 10月 13 18:47 rootfs.tar

What should I do next? Thank you again!

TPLuG9527, 2023-10-17:
> Thank you so much!! Seems to have a great progress.
>
> I run `git clone http://github.com/buildroot/buildroot` and `git clone https://github.com/litex-hub/linux-on-litex-vexriscv.git`
>
> and got
>
>
> then [...]

hey chat when following your steps it comes to these errors:
*
* Could not find Qt via pkg-config.
* Please install either Qt 4.8 or 5.x. and make sure it's in PKG_CONFIG_PATH
*
make[2]: *** No rule to make target '/home/chhooo/FPGA/buildroot/output/build/buildroot-config/.tmp_qtcheck', needed by '/home/chhooo/FPGA/buildroot/output/build/buildroot-config/qconf.o'.  Stop.
make[1]: *** [Makefile:958: /home/chhooo/FPGA/buildroot/output/build/buildroot-config/qconf] Error 2
make: *** [Makefile:82: _all] Error 2
so i wonder what's wrong

closed 2023-10-24 by 1487021

huangshaoxin, 2023-10-25:
hello!I use make to generate output.but there was a problem：

- I dont know how to solve it

huangshaoxin, 2023-10-25:
hello!I use make to generate output.but there was a problem：

I dont know how to solve it
![Uploading BG51XAT$`N`ATCY5A3O]X42.png…]()

huangshaoxin, 2023-10-25:
hello!I use make to generate output.but there was a problem：

GIT_DIR=/home/hsx/buildroot/dl/linux/git/.git git init . 
Reinitialized existing Git repository in /home/hsx/buildroot/dl/linux/git/.git/
GIT_DIR=/home/hsx/buildroot/dl/linux/git/.git git remote set-url origin 'https://github.com/Dolu1990/litex-linux.git' 
Fetching all references
GIT_DIR=/home/hsx/buildroot/dl/linux/git/.git git fetch origin 
remote: Enumerating objects: 9089940, done.
remote: Counting objects: 100% (603/603), done.
remote: Compressing objects: 100% (603/603), done.
fatal: the remote end hung up unexpectedly 1.01 GiB | 5.39 MiB/s   
fatal: early EOF
fatal: index-pack failed
@Dolu1990 
I dont know how to solve it