litex-hub/linux-on-litex-vexriscv #358: command make on Linux

BobMa0420, 2023-10-09:
Hi, I am running Linux on tang primer 20k and I'm wondering how I can use `make` on it. Thanks!

trabucayre, 2023-10-09:
You with a compile something on your tang primer 20k?
I'm not sure it's really relevant to compile with a CPU at 48MHz.
In fact buildroot does not provides tools to do native compile, it provides a cross-toolchain to build on your host code application.

BobMa0420, 2023-10-09:
> You with a compile something on your tang primer 20k? I'm not sure it's really relevant to compile with a CPU at 48MHz. In fact buildroot does not provides tools to do native compile, it provides a cross-toolchain to build on your host code application.

Thanks for your replying. Actually I am trying to run coremark on it but I am not sure whether it is a good idea to use `make` in the coremark folder.

trabucayre, 2023-10-09:
coremark seems present as buildroot's package (**coremark**).
So the best way is to use `make menuconfig` in buildroot root directory to select this package:
```
Target packages  --->
    Debugging, profiling and benchmark  --->
        [*] coremark
```
Followed by `make`.

And your new image must contain executable

BobMa0420, 2023-10-09:
> coremark seems present as buildroot's package (**coremark**). So the best way is to use `make menuconfig` in buildroot root directory to select this package:
>
> ```
> Target packages  ---> [...]

I use `make`  at /home/bobma/python-litex/buildroot and there is an error: package/Makefile.in:61: *** No C library enabled, this is not possible..  Stop.
make: *** [Makefile:82: _all] Error 2

BobMa0420, 2023-10-09:
> There is another issue here: seems buildroot has changed dependencies for glibc try to enable:
> 
> ```
> Target options  --->
>     [*] Atomic Instructions (A)
> ```

I run `make` and there is a `rootfs.tar` in `/home/bobma/python-litex/linux-on-litex-vexriscv/buildroot/output/images`, is that correct?

trabucayre, 2023-10-09:
You have to use the *rootfs.cpio*

BobMa0420, 2023-10-09:
> You have to use the _rootfs.cpio_

Thank you so much! But could you please give more details about it? Thank you so much!

TPLuG9527, 2023-10-16:
> I use `make` at /home/bobma/python-litex/buildroot and there is an error: package/Makefile.in:61: *** No C library enabled, this is not possible.. Stop. make: *** [Makefile:82: _all] Error 2

i input the same command at the same place but it comese errors:
mkdir -p /home/chhooo/FPGA/buildroot/output/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
    obj=/home/chhooo/FPGA/buildroot/output/build/buildroot-config -C support/kconfig -f Makefile.br qconf
*
* Could not find Qt via pkg-config.
* Please install either Qt 4.8 or 5.x. and make sure it's in PKG_CONFIG_PATH
*
make[2]: *** No rule to make target '/home/chhooo/FPGA/buildroot/output/build/buildroot-config/.tmp_qtcheck', needed by '/home/chhooo/FPGA/buildroot/output/build/buildroot-config/qconf.o'.  Stop.
make[1]: *** [Makefile:958: /home/chhooo/FPGA/buildroot/output/build/buildroot-config/qconf] Error 2
make: *** [Makefile:82: _all] Error 2

do u know how to solv it

closed 2023-10-19 by BobMa0420

RZQQQQ, 2023-10-30:
> > You have to use the _rootfs.cpio_
> 
> Thank you so much! But could you please give more details about it? Thank you so much!
Hello, I am also conducting coremark testing and have obtained the rootfs.cpio file after the same operation. Do you know what kind of operation to continue after obtaining the rootfs.cpio file?

RZQQQQ, 2023-10-30:
Hello, I am also conducting coremark testing and have obtained the rootfs.cpio file after the same operation. Do you know what kind of operation to continue after obtaining the rootfs.cpio file?