litex-hub/linux-on-litex-vexriscv #359: sipeed tang primer 20k freezes after the linux image is imported（高云20k开发版在导入linux镜像后出现卡死情况）

heisuo, 2023-10-09:
I used the following command`litex_term --images=images/boot.json /dev/ttyUSB1`to import the linux image into the sipeed tang primer 20k development release，but the serial port command line never appeared
我使用了如下命令`litex_term --images=images/boot.json /dev/ttyUSB1`将linux镜像导入sipeed tang primer 20k开发版，然而串口命令行始终没有出现
[image]
My soc parameters are as follows
[image]

allbert123, 2023-10-18:
Hi，请问这个问题解决了嘛？How to resolve the issue, If it is convenient, could you share the solution with me?

verilogzhou, 2023-10-21:
你可以用sd卡启动试一下

heisuo, 2023-10-21:
> Hi，请问这个问题解决了嘛？How to resolve the issue, If it is convenient, could you share the solution with me?

你好，我们尝试购买了一个新的开发板，这很好的解决了这个问题，所以这很有可能是开发板的问题
Hello, we tried to buy a new development board, which solved the problem very well, so it is very likely that the problem is the development board

allbert123, 2023-10-21:
> > Hi，请问这个问题解决了嘛？How to resolve the issue, If it is convenient, could you share the solution with me?
> 
> 你好，我们尝试购买了一个新的开发板，这很好的解决了这个问题，所以这很有可能是开发板的问题 Hello, we tried to buy a new development board, which solved the problem very well, so it is very likely that the problem is the development board

请问能添加其他联系方式详聊嘛，我对于这个项目还有一些其他的疑问想咨询，谢谢

heisuo, 2023-10-21:
> 你可以用sd卡启动试一下

很感谢你的回复！请问你是否知道哪里有关于配置该项目详细教程，我一直在寻找配置外设的方法，但是并没有找到这方面的信息。
Thank you very much for your reply! Do you know where there is a detailed tutorial on configuring this project, I have been looking for ways to configure peripherals, but have not found this information.

heisuo, 2023-10-21:
> 请问能添加其他联系方式详聊嘛，我对于这个项目还有一些其他的疑问想咨询，谢谢

欢迎交流！这是我的qq邮箱：2157848516@qq.com
Welcome to communicate! This is my qq email: 2157848516@qq.com

ShiXiaoTj, 2024-11-06:
> 你可以用sd卡启动试一下

请问一下，我想用sdcard启动，但是boot阶段没有识别sdcard，这是怎么回事呢？