litex-hub/linux-on-litex-vexriscv #360: when i use './sim.py',riscv64-unknown-elf-gcc raised "error: -march=rv32i2p0_ma: unsupported ISA substring '2p0_ma'"

bg51717, 2023-10-11:
I followed the instructions in the **README.me** step by step, but ran into an issue when executing **./sim.py.** .

First,after the following command is executed, Meson is called:

 `make -C /home/bs/linux-on-litex-vexriscv/build/sim/software/libc -f /home/bs/litex/litex/soc/software/libc/Makefile` 

From the log file, I noticed that Meson called 'riscv64-unknown-elf-gcc' with the specific command 

`riscv64-unknown-elf-gcc sanitycheckc.c -o sanitycheckc_cross.exe -Os -march=rv32i2p0_ma -mabi=ilp32 -D__vexriscv__ -DUART_POLLING -g3 -no-pie -fomit-frame-pointer -Wall -fno-builtin -fno-stack-protector -I/home/bs/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio -I/home/bs/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/include -I/home/bs/litex/litex/soc/software/libbase -I/home/bs/litex/litex/soc/software/include -I/home/bs/litex/litex/soc/software -I/home/bs/linux-on-litex-vexriscv/build/sim/software/include -I/home/bs/linux-on-litex-vexriscv/build/sim/software/include/../libc -I/home/bs/litex/litex/soc/cores/cpu/vexriscv_smp -fexceptions -Wpragmas -D_FILE_OFFSET_BITS=64 -c`. 

The error message was 

`-march=rv32i2p0_ma: unsupported ISA substring '2p0_ma'`.

 How can I resolve this issue? Thank you, and I can provide the specific meson-log.txt if needed.

jeremy-heath, 2023-10-14:
I had the same error which went away when I reinstalled the Litex+Linux-on-litex-vexriscv environment and compiler.

a workaround is to determine the compiler's known substrings and modify the meson config to use a known substring in place of 2p0_ma.   It's been too long to remember the exact steps.

bg51717, 2023-10-22:
I have solved this problem by this [cc1: error, Compiler riscv64-unknown-elf-gcc can not compile programs #10625](https://github.com/mesonbuild/meson/issues/10625).

closed 2023-11-15 by bg51717