litex-hub/linux-on-litex-vexriscv #362: ULX3S: Quad-core system hangs at FPGA manager framework

stettberger, 2023-10-19:
TL;DR: Multi-Core System does not boot Linux. Framebuffer makes everything worse

Hello! I have the goal to build a quad-core RISC-V system that I load
onto an ULX3S Board (ECP-85F). In the long term, I want to use this
setup to give university-level lecture on multi-core operating system
construction.

At this point, I want to boot a Linux system on a synthesized quadcore
to test the gateware, before I start porting my own software there,
hunting gateware-level bugs. Also: I'm aware of other softcores
(Rocket, NaxRiscv, PULP) but my current goal is to make it running
with the VexRiscv.

Problem
-------

I am not able to get a multi-core system with > 2 cores
running as it hangs before the user-land fully starts. The last line
on the litex_terminal is:

```
[  268.081505] FPGA manager framework
```

**Suspicion**: I suspect two problems here: There is some multicore
problem _and_ the litex framebuffer yields unfavorable logic for the
ulx3s.

Overview
--------

I got a dual core system to boot with `--board=ulx3s --cpu-count=2
--with-wishbone-memory --device LFE5U-85F`. However, I had to reduce
the system clock to 25Mhz and disable the "framebuffer". Besides from
that, I tried the following configurations:

- [good] 2 cores, whishbone-memory, 25 Mhz
- [bad] 2 cores, wishbone-memory, 25 Mhz, with framebuffer: Timing warning, Boot hangs at FPGA manager framework
- [bad] 4 cores, whishbone memory, 25Mhz, No timing warning, boot hangs at "FPGA manager framework", boot is super slow (200 seconds till hang)
- [bad] 4 cores, No timing warning, boot hangs at "FPGA manager framework", 25 Mhz

  Also, with the framebuffer enabled, I had tremendously long nextpnr runtime (> 1.5 h).

Detail for 4 cores
------------------

As an example, I will provide the some logs for my last attempt (4
cores, 25 Mhz, no framebuffer, no whishbone memory).

For this test, I built the buildroot by myself. but I also tried the
downloaded/pre-built buildroot before, and it gave the same problem.

Version Info:

```shell
$  yosys -V
Yosys 0.34+23 (git sha1 59fbee400, clang 10.0.0-4ubuntu1 -fPIC -Os)
$  nextpnr-ecp5 -V
"nextpnr-ecp5" -- Next Generation Place and Route (Version nextpnr-0.6-119-g5a2eff21)
$  ecppack --version
Project Trellis - Open Source Tools for ECP5 FPGAs
Version 1.4-64-g36c615d
/home/stettberger/ulx3s/oss-cad-suite/libexec/ecppack: ECP5 bitstream packer
$  git log --oneline | head -n1                      :(
108c020 buildroot/configs/litex_vexriscv_xx: fixed missing C library by replacing BR2_RISCV_ISA_CUSTOM to BR2_RISCV_ISA (buildroot's commit cbd91e89e4)
$ git diff
diff --git a/make.py b/make.py
index 31ebb38..3511214 100755
--- a/make.py
+++ b/make.py
[... 534 lines cut ...]
                  / __/  |/  / _ \
                 _\ \/ /|_/ / ___/
                /___/_/  /_/_/
  32-bit RISC-V Linux running on LiteX / VexRiscv-SMP.

login[76]: root login on 'console'
root@buildroot:~# cat /proc/cpuinit
cat: can't open '/proc/cpuinit': No such file or directory
root@buildroot:~# cat /proc/cpuinfo 
processor	: 0
hart		: 0
isa		: rv32ima
mmu		: sv32
mvendorid	: 0x0
marchid		: 0x0
mimpid		: 0x0

processor	: 1
hart		: 1
isa		: rv32ima
mmu		: sv32
mvendorid	: 0x0
marchid		: 0x0
mimpid		: 0x0
```

Dolu1990, 2023-10-19:
Hi,

One very possible cause to your slow boot issue is that the linux kernel tick too fast for a too slow system.
By default the linux tick rate is 250hz which is fast for a 25 Mhz single core adding on that 3 other cores generating traffic (and ticking as well), + the slow wishbone memory, it is very probable that the quad core is only using time ticking and can't do any forward progress :)

There is a config in the in the kernel that you can enable : CONFIG_HZ_100
Let's me know how things goes :D

stettberger, 2023-10-20:
@Dolu1990 CONFIG_HZ_100 really helped a lot. I'm now able to build and boot a quadcore system (see below).  It was not enough to only enable 100 Hz ticks, but I also had to increase the clock frequency to 40 Mhz (with 45 Mhz in reach):

```
Info: Max frequency for clock '$glbnet$crg_clkout0': 45.77 MHz (PASS at 40.00 MHz)
```

For this repo:
- Would it make sense to always set HZ_100?
- Should the ULX3S class in `make.py` decrease the clockspeed for > 1 core?

I will now investigate on the framebuffer

Dolu1990, 2023-10-20:
Nice :)

> Would it make sense to always set HZ_100?

Yes, i think you are right, that would be a more portable default.

Also, if possible avoid --with-wishbone-memory as it realy hit the memory bandwidth between the SoC and the SDRAM

The framebuffer is also probably starving data, i had that when for instance i was running digilent video at 50 Mhz, just got blackscreen. Not sure what is your actual resolution ?

stettberger, 2023-10-20:
Currently, I have not changed the default settings of framebuffer at all. Therefore, I cannot tell about the resolution that the framebuffer has. In the end, I would be more than happy to have a 80x25 text terminal (as in former times with CGA and Hercules cards).

stettberger, 2023-10-20:
Ok. So when using the framebuffer, the system seems to be memory starving on the ulx3s. However, If I enable the `"video_terminal"`, I can synthesize a quad-core with HDMI out. And while it is working, I get some timing warning:

```
Warning: Max frequency for clock '$glbnet$ecp5pll1_clkout1': 97.14 MHz (FAIL at 125.00 MHz)
Info: Max frequency for clock '$glbnet$ecp5pll1_clkout0': 70.81 MHz (PASS at 25.00 MHz)
Info: Max frequency for clock '$glbnet$ecp5pll0_clkout0': 48.69 MHz (PASS at 40.00 MHz)

Info: Clock '$glbnet$ecp5pll0_clkout1' has no interior paths

Info: Max delay <async>                          -> posedge $glbnet$ecp5pll0_clkout0: 12.49 ns
Info: Max delay <async>                          -> posedge $glbnet$ecp5pll1_clkout0: 8.66 ns
Info: Max delay <async>                          -> posedge $glbnet$ecp5pll1_clkout1: 8.96 ns
Info: Max delay posedge $glbnet$ecp5pll0_clkout0 -> <async>                         : 5.25 ns
Info: Max delay posedge $glbnet$ecp5pll0_clkout0 -> posedge $glbnet$ecp5pll1_clkout0: 3.87 ns
Info: Max delay posedge $glbnet$ecp5pll0_clkout0 -> posedge $glbnet$ecp5pll1_clkout1: 4.16 ns
Info: Max delay posedge $glbnet$ecp5pll1_clkout0 -> posedge $glbnet$ecp5pll0_clkout0: 1.46 ns
Info: Max delay posedge $glbnet$ecp5pll1_clkout0 -> posedge $glbnet$ecp5pll1_clkout1: 1.24 ns
Info: Max delay posedge $glbnet$ecp5pll1_clkout1 -> posedge $glbnet$ecp5pll1_clkout0: 1.48 ns
```

It does not seem to be a problem, however, I'm not sure what it means that the pll1_clkout1 is less than 125 Mhz.

Further, the video_terminal is not to useful with Linux using the uart for its login shell.

**My Next Step**: I also think that it should be doable to modify "VideoTerminal" so that it (a) does not sniff the uart sink for its data, but that it (b) exposes the video and the font memory via wishbone.  Thereby, I should become able to emulate a good old text-mode graphic card.

@Dolu1990 Would it make sense (in the multi-core mode) to (a) Disable framebuffer for the ULX3S Board (b) reduce the clock frequency?

Dolu1990, 2023-10-20:
Hi,

> pll1_clkout 

I think that is to drive the 640x480@75Hz

> Further, the video_terminal is not to useful with Linux using the uart for its login shell.

See https://github.com/SpinalHDL/NaxSoftware/tree/main/debian_litex#boot-console

So console=tty1  in the linux bootcmd may fix it ?

If you want, you can try to get the USB host working. (there is a OHCI controller in litex)

> Would it make sense (in the multi-core mode) to (a) Disable framebuffer for the ULX3S Board (b) reduce the clock frequency?

I'm not sure. There is so many possibility / use cases. I would say people need to customiz for their use case.