litex-hub/linux-on-litex-vexriscv #365: No serial on Genesys 2 board

newinnovations, 2023-10-22:
I am working with a Genesys 2 board and succeeded in creating the gateware using:

```text
./make.py --board genesys2
```

Creating the bitstream and uploading it to the board using Vivado also went ok, and i see the chaser animation on the leds.

But I don't see any activity on the serial port. Looking at the generated verilog files, I noticed that there is no serial connection to the core:

```verilog
module genesys2 (
    input  wire          clk200_n,
    (* dont_touch = "true" *)
    input  wire          clk200_p,
    input  wire          cpu_reset_n,
    output wire   [14:0] ddram_a,
    output wire    [2:0] ddram_ba,
    output wire          ddram_cas_n,
    output wire          ddram_cke,
    output wire          ddram_clk_n,
    output wire          ddram_clk_p,
    output wire          ddram_cs_n,
    output wire    [3:0] ddram_dm,
    inout  wire   [31:0] ddram_dq,
    inout  wire    [3:0] ddram_dqs_n,
    inout  wire    [3:0] ddram_dqs_p,
    output wire          ddram_odt,
    output wire          ddram_ras_n,
    output wire          ddram_reset_n,
    output wire          ddram_we_n,
    (* dont_touch = "true" *)
    input  wire          eth_clocks_rx,
    output wire          eth_clocks_tx,
    input  wire          eth_int_n,
    output wire          eth_mdc,
    inout  wire          eth_mdio,
    output wire          eth_rst_n,
    input  wire          eth_rx_ctl,
    input  wire    [3:0] eth_rx_data,
    output wire          eth_tx_ctl,
    output wire    [3:0] eth_tx_data,
    input  wire          sdcard_cd,
    output wire          sdcard_clk,
    inout  wire          sdcard_cmd,
    inout  wire    [3:0] sdcard_data,
    output wire          sdcard_rst,
    inout  wire    [7:0] usb_fifo_data,
    input  wire          usb_fifo_oe_n,
    output reg           usb_fifo_rd_n,
    input  wire          usb_fifo_rxf_n,
    input  wire          usb_fifo_siwua,
    input  wire          usb_fifo_txe_n,
    output reg           usb_fifo_wr_n,
    output wire          user_led0,
    output wire          user_led1,
    output wire          user_led2,
    output wire          user_led3,
    output wire          user_led4,
    output wire          user_led5,
    output wire          user_led6,
    output wire          user_led7
);
```

Looking at `make.py`, I see that a lot of boards get `serial` added to `soc_capabilities` during board class initialization. But the Genesys2 does not. Adding it manually made no difference.

Am I missing something here? Because I saw several people here posting serial logs from Genesys 2 boards.

Your help is greatly appreciated.

tparng, 2023-10-25:
I am also working with Genesys2 board and have similiar issues.  I can also upload bitstream to the Genesys2 board using Vivado. But if try to load the bitstream with command "$ ./make.py --board=XXYY --cpu-count=X --load", the command runs but nothing happened on the Genesys2 board (proramming done LED not light up and no chasing LEDs animation).  Besides, the Load Linux Images over Serial command "$ litex_term --images=images/boot.json /dev/ttyUSB0" did not not work either (It just freezed there, nothing happened.).  It seems that the JTAG or  litex_term conection is OK but  bitstream or linux image programming is not done successfully.

tparng, 2023-11-17:
@newinnovations, 

I encountered  the same issue as you. After digging into the code, I found the serial io pads are mistakenly (due to bugs or other reasons) taken by usb_fifo.  Therefore I fixed the bug by adding "serial” (this may be not necessary)and remove “usb_fifo” in class Genesys2(Board) in make.py :

# Genesys2 support ---------------------------------------------------------------------------------

class Genesys2(Board):
	def __init__(self):
    	from litex_boards.targets import digilent_genesys2
    	Board.__init__(self, digilent_genesys2.BaseSoC, soc_capabilities={
        	# Communication
        	"serial",
        	#"usb_fifo",
        	"ethernet",
        	# Storage
        	"sdcard",
   	})
Now, I can do serial boot successfully for linux-on-litex-vexriscv.
You can take a try!

BTW, if you also want to try booting on Ethernet, remember to have your tftp server configured to have a BLOCKSIZE of 1024 as ~/litex/litex/soc/software/libliteeth/tftp.c has it pre-defined to be 1024:
#define    BLOCK_SIZE    1024 /* block size in bytes 512 or 1024*/

Tparng

adambagley, 2025-02-12:
I connected 2 microUSB cables between my genesys2 and PC, one on the board port labeled JTAG and the other on the board port labeled UART. On my PC it ended up being Vivado auto-detects the JTAG just fine for programming. And then for regular Litex my serial port works for /dev/ttyUSB0, but when I do linux-on-litex-vexriscv I need to use /dev/ttyUSB1 and it works fine for me. Therefore I don't think 'usb_fifo' is a bug, it's just a different serial mechanism on a different USB port. Try changing which /dev/ttyUSBX you use for litex_term or minicom or however you're accessing serial port