litex-hub/linux-on-litex-vexriscv #366: upload bitstream to Genesys2 board done but chasing LEDs animation doesn't start

tparng, 2023-10-27:
I am trying to test linux-on-litex-vexriscv on Genesys2 board. When I load the bitstream with command "$ ./make.py --board=genesys2 --cpu-count=4 --load", the command runs but nothing happened on the Genesys2 board. However, after I launch Vivado and open a project and click Open Hardware Manager => Open Target =>Auto Connect, the "Programming Done LED" lights up and the chasing LEDs animation start to run.  It seems that "./make.py --board=genesys2 --cpu-count=4 --load" can finish bitstream programming but does not activate the hardware to start running.  Is there anything wrong with the configuration/settings ?
Thanks,

TParng

tparng, 2023-10-27:
I found the solution: reduce the "adapter_khz"  (in linux-on-litex-vexriscv/prog/openocd_genesys2.cfg) from 25000 down to  4000 ~ 1000.  Then it works fine.

enjoy-digital, 2024-07-09:
Hi @tparng,

sorry for the delay and thanks for sharing your finding/feedback. Configuration has been working fine here are 25MHz on the Genesys2. The issue is probably related to your USB cable. I'm closing this but will think about this if I encouter this issue while using the Genesys2 and will then lower frequency if I reproduce.

closed 2024-07-09 by enjoy-digital