litex-hub/linux-on-litex-vexriscv #367: Is there any way to run python programs in linux-on-litex-vexriscv

allrighteveryday, 2023-10-27:


Dolu1990, 2023-10-27:
you need to add python in buildroot
Look at the readme about how to compile buildroot, but before compileing it you can do   "make xconfig" this will open a gui, there you can do ctrl + f    python

allrighteveryday, 2023-10-28:
> you need to add python in buildroot Look at the readme about how to compile buildroot, but before compileing it you can do "make xconfig" this will open a gui, there you can do ctrl + f python

thanks for guidance.it does work!
and is there any way to see the power performance area of the vexriscv processor?

Dolu1990, 2023-10-30:
so far, i would say you can check the synthesis results, but the values are a mix for the whole soc, so not so much Vex specific

allrighteveryday, 2023-11-06:
> you need to add python in buildroot Look at the readme about how to compile buildroot, but before compileing it you can do "make xconfig" this will open a gui, there you can do ctrl + f python

micropython is ok.but when I select python3, it doesn't work. or Do we have to check some dependency?
when enter python3 in linux , the errors is 
`"Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding"`

by the way ,I really wonder why I can't execute a C programs by gcc in the linux :(

any help apprecited :)

allrighteveryday, 2023-11-06:
> you need to add python in buildroot Look at the readme about how to compile buildroot, but before compileing it you can do "make xconfig" this will open a gui, there you can do ctrl + f python

micropython is ok.but when I select python3, it doesn't work. or Do we have to check some dependency?
when enter python3 in linux , the error is
`Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding`

by the way ,I really wonder why I can't execute a C programs by gcc in the linux :(

any help apprecited :)

enjoy-digital, 2024-07-09:
Hi @allrighteveryday,

you can enable Python3 by adding the package to your buildroot defconfig:` BR2_PACKAGE_PYTHON3=y`

closed 2024-07-09 by enjoy-digital