litex-hub/linux-on-litex-vexriscv #368: Problems with HDMI display on Linux

mmx-114514, 2023-10-30:
How to display via HDMI on linux system please? Is it by mounting the hdmi peripheral via liitex and then selecting the HDMI display related compilation options on buildroot, I tried to do this but it doesn't show up.

Dolu1990, 2023-10-30:
Do you have it in the DTS ?

mmx-114514, 2023-11-01:
It doesn't seem to be there. I selected hdmi via litex and I was able to see its constraints on the pins in .cst, but I can't find any configuration for hdmi in dts. Do I have to write my own documentation for this?

Dolu1990, 2023-11-01:
ahhh curious, and is the HDMI displaying some random static noise when you boot ? Thing is for Digilent nexys video for example, if i remember well, the framebuffer was included in the dts by default

mmx-114514, 2023-11-01:
Sorry, there is no output when HDMI is activated. I'm using Sipeed_tang_20k and the output dts file has nothing related to hdmi and hdmi has no output. I double-checked make.py in linux-on-litex-vexriscv to make sure that video_terminal was added, and re-executed make.py, and the .dts file still had no hdmi related configuration after compilation.

Dolu1990, 2023-11-02:
Ahhh, i don't know much about that side of things. no hdmi output mean no hdmi signal at all ?

mmx-114514, 2023-11-02:
Yes. I was dumbfounded by myself that I didn't add the framebuffer xD. so far I solved the problem now I have the output of the linux terminal info. Thanks buddy.

code11killer, 2023-11-03:
> Yes. I forgot to add the framebuffer xD. so far I solved the problem now I have the output of the linux terminal info. Thanks buddy.

hello，I want to ask how to solve the problem. I meet the problem ，too

mmx-114514, 2023-11-03:
> code11killer
Add framebuffer to make.py

code11killer, 2023-11-04:
> > code11killer
> > Add framebuffer to make.py

I add it to make.py. but I don't know how to output the video in my screen. beacuse I can't use the xrandr. Is there any other way? please

code11killer, 2023-11-04:
> > code11killer
> > Add framebuffer to make.py

and after I add it，I also can't see the video in DTS. If it's convenient, can you add a contact information to talk about it?

mmx-114514, 2023-11-04:
qq 1364449373

mmx-114514, 2023-11-04:
> 啊，我对那方面的事情了解不多。没有HDMI输出意味着根本没有HDMI信号？

Hello, I am experiencing a new problem, after introducing frame buffer output HDMI, the cpu performance is severely degraded, I have enabled --with-coherent-dma but the effect is not improved, I don't know how coherent-dma works here.

mmx-114514, 2023-11-04:
> Ahhh, i don't know much about that side of things. no hdmi output mean no hdmi signal at all ?

Hello, I am experiencing a new problem, after introducing frame buffer output HDMI, the cpu performance is severely degraded, I have enabled --with-coherent-dma but the effect is not improved, I don't know how coherent-dma works here.

mmx-114514, 2023-11-04:
> Ahhh, i don't know much about that side of things. no hdmi output mean no hdmi signal at all ?

By the way, I currently implemented a terminal output text via HDMI on the display, and I would like to be able to display images, after adding the fbv related libraries in buildroot, cat /dev/urandom > /dev/fb0 can output garbled code normally with uart and HDMI, but I can't display the images with fbv (image), the error Displayed as: ?25lopen /dev/fb0: No such file or directory. and I can't find anything about framebuffer in DTS. I would like to know what I can do to make linux access the framebuffer for image display.