litex-hub/linux-on-litex-vexriscv #369: Using Litescope on ULX3s

LucasKl, 2023-11-09:
Hi,
I'm trying to add a Litescope analyzer to the ULX3S configuration for use on an ULX3S 85 r3.
For this, I'm following [this](https://github.com/enjoy-digital/litex/wiki/Use-LiteScope-To-Debug-A-SoC) tutorial from the Litex wiki.
As a first step, I am trying to set up a UARTBone bridge as described [here](https://github.com/enjoy-digital/litex/wiki/Use-Host-Bridge-to-control-debug-a-SoC#add-a-uartbone-bridge-to-your-soc-serial-console).

I updated the ULX3S board configuration to this (framebuffer commented to reduce synth time, I got the same result with it enabled):
```python
class ULX3S(Board):
    soc_kwargs = {"l2_size" : 2048} # Use Wishbone and L2 for memory accesses.
    def __init__(self):
        from litex_boards.targets import radiona_ulx3s
        Board.__init__(self, radiona_ulx3s.BaseSoC, soc_capabilities={
            # Communication
            "serial",
            # Storage
            "spisdcard",
            "crossover",

            # Video,
            # "framebuffer",
        })
```
I build, upload, and start the server using the following commands:
```
./make.py --board=ULX3S --device=LFE5U-85F --cpu-count=1 --build
./make.py --board=ULX3S --device=LFE5U-85F --cpu-count=1 --load
$ litex_server --uart --uart-port=/dev/ttyUSB0 --debug
[CommUART] port: /dev/ttyUSB0 / baudrate: 115200 / tcp port: 1234
```

Without the crossover I was able to boot into Linux from serial and from the SD card following the instruction (however I had to change "sdcard" to "spisdcard" which I guess is due to a change in the new revision). The serial is connected to the USB1 UART.

With the crossover I get no response from the board. It is listed as `/dev/ttyUSB0` but if I connect to it via *litex_term* or *picocom* I get no response. Running `litex_term crossover --csr-csv build/ulx3s/csr.csv` results in a timeout error.

```
$ litex_term crossover --csr-csv build/ulx3s/csr.csv --safe
Exception in thread Thread-2 (crossover2pty):
Traceback (most recent call last):
  File "/usr/lib64/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/klemmer/Desktop/litex/litex/tools/litex_term.py", line 121, in crossover2pty
    if self.rxfull.read():
       ^^^^^^^^^^^^^^^^^^
  File "/home/klemmer/Desktop/litex/litex/tools/remote/csr_builder.py", line 40, in read
    datas = self.readfn(self.addr, length=self.length)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/klemmer/Desktop/litex/litex/tools/litex_client.py", line 88, in read
    init       = self.receive_packet(self.socket, addr_size)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/klemmer/Desktop/litex/litex/tools/remote/etherbone.py", line 406, in receive_packet
    chunk = socket.recv(header_length - len(packet))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TimeoutError: timed out
```
On the server side, I get `Connected with 127.0.0.1:38120` after starting *litex_term*.

Am I missing something that has to be set up before?

Anyway, thanks for creating and maintaining this repository/litex. It is super cool that this works on so many board so easily!

enjoy-digital, 2024-07-09:
Hello @LucasKl,

upload of linux images over UART crossover will be very slow and hasn't been tested. To use LiteScope in your design, I would just recommend adding a second UART to your design and use `add_uartbone(...)` of LiteX.

closed 2024-07-09 by enjoy-digital

LucasKl, 2024-07-09:
All right, thank you!