litex-hub/linux-on-litex-vexriscv #374: How can I capture the waveform of the portion where the C program was executed in the simulation using sim.py?

nohahanon, 2023-12-08:
Hi there!

I was able to execute the C program in sim.py and now I want to observe the waveform of that part. When I execute it like 
```
./sim.py --trace --trace-fst
```
it creates a waveform file that fills up the disk because it's so large(>200GB). Currently, I'm investigating parts little by little by running
```
./sim.py --trace --trace-fst --trace-start 400000000000000 --trace-end 450000000000000
```
 but it seems like it'll take a long time to cover everything.
Do you have any good ideas?

Thanks in advance!!

Dolu1990, 2023-12-11:
Hi,

Getting waveform late in stuff which boot linux is a real paine ^^
When using Litex, i do the same as you.
Also i think that once i manualy modified the litex sampling code to periodicaly take snapshot of a few 100's cycles every 100000 cycles (usefull to see when things get stuck)

But in general, what i do when i have to debug the CPU not running properly, is to run some lock-step simulation of the CPU / SoC alone (but that's without the litex layer, only SpinalHDL stuff)

I don't know if litex has a way to trigger wave capture by writing into a peripheral region, that could solve your requirements in a  quite flexible way.

nohahanon, 2023-12-15:
Thanks for your replying! I'll search that way!

closed 2023-12-15 by nohahanon