litex-hub/linux-on-litex-vexriscv #376: Can add official support for Sipeed tang primer board (using Anlogic’s FPGA EG4S20BG256)？

hai4john, 2023-12-25:
I saw a blog (https://www.luffca.com/2022/10/linux-tang-primer/) Running Dual-Core RISC-V Linux on Sipeed tang primer board. 
Can add official support for Sipeed tang primer board (using Anlogic’s FPGA EG4S20BG256)？

hai4john, 2023-12-25:
https://github.com/ReinForce-II/anlogic_eg4s_sdram_controller
The link for reference

enjoy-digital, 2024-07-09:
Thanks @hai4john for the link, sadly the project provide some information but does not seems to be open or available. I just added a link to it to the LiteX Wiki: https://github.com/enjoy-digital/litex/wiki/Tutorials-Resources but unless someone wants to re-do the work or find it, share the support, I don't think we'll have time to add support for it.

closed 2024-07-09 by enjoy-digital