litex-hub/linux-on-litex-vexriscv #377: Trying to port for TUL PYNQ Z2 Error [KeyError: 'main_ram']

ZeroX29a, 2023-12-29:
To the make file i have added this class to suprt the pynq z2
```
class pynqz2(Board):
    soc_kwargs = {"uart_name": "serial"}
    def __init__(self):
        from litex_boards.targets import tul_pynq_z2
        Board.__init__(self, tul_pynq_z2.BaseSoC, soc_capabilities={
            "serial",
            # Storage
            "spiflash",
            # GPIOs
            "leds",
            "switches",
            "xadc",
            # 7-Series specific
            "mmcm",
            "icap_bitstream",
        })

```
and ran the command  
``` ./make.py --board pynqz2 --build ``` 
Is there anything to be added?
please help
the error was
```
make: Leaving directory '/home/litex/linux-on-litex-vexriscv/build/pynqz2/software/liblitesata'
make: Entering directory '/home/litex/linux-on-litex-vexriscv/build/pynqz2/software/bios'
 CC       boot.o
 CC       cmd_bios.o
 CC       cmd_mem.o
 CC       cmd_boot.o
 CC       cmd_i2c.o
 CC       cmd_spiflash.o
 CC       cmd_litedram.o
 CC       cmd_liteeth.o
 CC       cmd_litesdcard.o
 CC       cmd_litesata.o
 CC       sim_debug.o
 CC       main.o
 CC       crt0.o
 CC       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.crcfbigen bios.bin --little
python3 -m litex.soc.software.memusage bios.elf /home/litex/linux-on-litex-vexriscv/build/pynqz2/software/bios/../include/generated/regions.ld riscv64-unknown-elf

ROM usage: 19.28KiB     (30.12%)
RAM usage: 0.05KiB      (0.78%)

rm crt0.o
make: Leaving directory '/home/litex/linux-on-litex-vexriscv/build/pynqz2/software/bios'
INFO:SoC:Initializing ROM rom with contents (Size: 0x4d2c).
INFO:SoC:Auto-Resizing ROM rom from 0x10000 to 0x4d2c.
Traceback (most recent call last):
  File "./make.py", line 1012, in <module>
    main()
  File "./make.py", line 988, in main
    soc.generate_dts(board_name)
  File "/home/litex/linux-on-litex-vexriscv/soc_linux.py", line 133, in generate_dts
    dts_content = generate_dts(json.load(json_file), polling=False)
  File "/home/litex/litex_installation/litex/litex/tools/litex_json2dts_linux.py", line 84, in generate_dts
    linux_initrd_start = d["memories"]["main_ram"]["base"] + initrd_start,
KeyError: 'main_ram'
```

ZeroX29a, 2023-12-29:
gathered some insight about the issue. arty board had successfully built the image, up on inspection i found arty board has DDR3 RAM enabled with "main_ram"
```
        # DDR3 SDRAM -------------------------------------------------------------------------------
        if not self.integrated_main_ram_size:
            self.ddrphy = s7ddrphy.A7DDRPHY(platform.request("ddram"),
                memtype        = "DDR3",
                nphases        = 4,
                sys_clk_freq   = sys_clk_freq)
            self.add_sdram("sdram",
                phy           = self.ddrphy,
                module        = MT41K128M16(sys_clk_freq, "1:4"),
                l2_cache_size = kwargs.get("l2_size", 8192)
            )
```
source: /litex_boards/targets/diligent_arty.py
mot sure if it is connected with the issue but
 can someone please help me with how to initialize DDR3 for pynq z2

ZeroX29a, 2023-12-29:
From the datasheets, DDR is directly connected to PS and not PL so

ZeroX29a, 2023-12-29:
i have managed to make it work by adding  ```  kwargs["integrated_main_ram_size"] = 0x10000 ```
in ``` /litex_boards/targets/tul_pynq_z2.py ```

now i get new error
```
INFO: [Vivado 12-3199] DRC finished with 0 Errors, 56 Warnings
INFO: [Vivado 12-3200] Please refer to the DRC report (report_drc) for more information.
INFO: [Designutils 20-2272] Running write_bitstream with 8 threads.
Loading data files...
Loading site data...
Loading route data...
Processing options...
Creating bitmap...
Creating bitstream...
Writing bitstream ./pynqz2.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-1876] WebTalk data collection is mandatory when using a ULT device. To see the specific WebTalk data collected for your design, open the usage_statistics_webtalk.html or usage_statistics_webtalk.xml file in the implementation directory.                                            
INFO: [Common 17-83] Releasing license: Implementation
9 Infos, 56 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:12 ; elapsed = 00:00:07 . Memory (MB): peak = 3194.102 ; gain = 46.617 ; free physical = 3274 ; free virtual = 9200
# quit
INFO: [Common 17-206] Exiting Vivado at Sat Dec 30 02:03:31 2023...
build/pynqz2/pynqz2.dts:164.34-175.19: Warning (unit_address_vs_reg): /soc/clk@f0003000/CLKOUT0: node has a reg or ranges property, but no unit name
build/pynqz2/pynqz2.dts:177.34-188.19: Warning (unit_address_vs_reg): /soc/clk@f0003000/CLKOUT1: node has a reg or ranges property, but no unit name

```

ZeroX29a, 2023-12-30:
This turnsout to be a working bitstream
but the ram size turned out to be just 6KiB so cant run the linux and increassing the integrated_main_ram_size results in this error 
```
---------------------------------------------------------------------------------
Loading part: xc7z020clg400-1
---------------------------------------------------------------------------------
Finished Loading Part and Timing Information : Time (s): cpu = 00:00:20 ; elapsed = 00:00:22 . Memory (MB): peak = 2699.109 ; gain = 1076.430 ; free physical = 3341 ; free virtual = 9325
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Start Applying 'set_property' XDC Constraints
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Finished applying 'set_property' XDC Constraints : Time (s): cpu = 00:00:20 ; elapsed = 00:00:22 . Memory (MB): peak = 2699.109 ; gain = 1076.430 ; free physical = 3341 ; free virtual = 9325
---------------------------------------------------------------------------------
WARNING: [Synth 8-3936] Found unconnected internal register 'MmuPlugin_shared_pteBuffer_PPN1_reg' and it is trimmed from '12' to '10' bits. [/home/litex/litex_installation/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm_Hb1.v:11670]                                                                            
WARNING: [Synth 8-3936] Found unconnected internal register 'memory_to_writeBack_INSTRUCTION_reg' and it is trimmed from '32' to '30' bits. [/home/litex/litex_installation/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm_Hb1.v:8361]                                                                             
WARNING: [Synth 8-3936] Found unconnected internal register 'execute_to_memory_INSTRUCTION_reg' and it is trimmed from '32' to '30' bits. [/home/litex/litex_installation/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm_Hb1.v:8142]                                                                               
INFO: [Synth 8-802] inferred FSM for state register 'IBusCachedPlugin_injector_port_state_reg' in module 'VexRiscv'                                                                                   
WARNING: [Synth 8-3936] Found unconnected internal register 'storage_1_dat1_reg' and it is trimmed from '10' to '8' bits. [/home/litex/linux-on-litex-vexriscv/build/pynqz2/gateware/pynqz2.v:2852]   
WARNING: [Synth 8-3936] Found unconnected internal register 'storage_dat1_reg' and it is trimmed from '10' to '8' bits. [/home/litex/linux-on-litex-vexriscv/build/pynqz2/gateware/pynqz2.v:2831]     
---------------------------------------------------------------------------------------------------
                   State |                     New Encoding |                Previous Encoding 
---------------------------------------------------------------------------------------------------
                  iSTATE |                              000 |                              000
                 iSTATE0 |                              001 |                              001
                 iSTATE1 |                              010 |                              010
                 iSTATE2 |                              011 |                              011
                 iSTATE3 |                              100 |                              100
---------------------------------------------------------------------------------------------------
INFO: [Synth 8-3354] encoded FSM with state register 'IBusCachedPlugin_injector_port_state_reg' using encoding 'sequential' in module 'VexRiscv'                                                      
WARNING: [Synth 8-6841] Block RAM (main_ram_reg) originally specified as a Byte Wide Write Enable RAM cannot take advantage of ByteWide feature and is implemented with single write enable per RAM due to following reason.                                                                             
(address width (21) is more than optimal threshold of 12. Implementing using BWWE will require more logic and timing would be suboptimal. Please use attribute ram_decomp = power if BWWE is desired.)
---------------------------------------------------------------------------------
Finished RTL Optimization Phase 2 : Time (s): cpu = 00:00:25 ; elapsed = 00:00:27 . Memory (MB): peak = 2699.109 ; gain = 1076.430 ; free physical = 3336 ; free virtual = 9323
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Start RTL Component Statistics 
---------------------------------------------------------------------------------
Detailed RTL Component Info : 
+---Adders : 
           2 Input   64 Bit       Adders := 1     
           3 Input   52 Bit       Adders := 1     
           2 Input   33 Bit       Adders := 3     
           3 Input   33 Bit       Adders := 1     
           3 Input   32 Bit       Adders := 2     
           2 Input   32 Bit       Adders := 5     
           2 Input   12 Bit       Adders := 1     
[... cut ...]
  File "/home/litex/litex_installation/litex/litex/build/xilinx/vivado.py", line 141, in build
    return GenericToolchain.build(self, platform, fragment, **kwargs)
  File "/home/litex/litex_installation/litex/litex/build/generic_toolchain.py", line 123, in build
    self.run_script(script)
  File "/home/litex/litex_installation/litex/litex/build/xilinx/vivado.py", line 401, in run_script
    raise OSError("Error occured during Vivado's script execution.")
OSError: Error occured during Vivado's script execution.
```

ratzupaltuff, 2024-01-14:
I can't help here either but I wanted to add a related issue here, maybe it is interesting for you: 
https://github.com/litex-hub/linux-on-litex-vexriscv/issues/339

ZeroX29a, 2024-01-20:
@ratzupaltuff  Thank you, it helped, sadly I think atm its not possible to connect the ddr from PS to PL.

closed 2024-01-20 by ZeroX29a