litex-hub/linux-on-litex-vexriscv #379: sim.py: Could not initialize class sbt.internal.parser.SbtParser$

roby2014, 2024-04-01:
Hello. Im trying to build and simulate, however, I'm getting this error with `sbt`:
```
(.venv) [roby@thonkpad linux-on-litex-vexriscv]$ ./sim.py 
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2024-04-01 19:50:29)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : SIM.
INFO:SoC:System clock: 100.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU vexriscv_smp added.
INFO:SoC:CPU vexriscv_smp adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode:  RW, Cached: False, Linker: False.
INFO:SoC:CPU vexriscv_smp overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU vexriscv_smp setting reset address to 0x00000000.
INFO:SoC:CPU vexriscv_smp adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoC:CPU vexriscv_smp adding Interrupt(s).
INFO:SoCIRQHandler:noirq IRQ added at Location 0.
INFO:SoC:CPU vexriscv_smp adding SoC components.
INFO:SoCCSRHandler:uart CSR added at Location 2.
INFO:SoCCSRHandler:timer0 CSR added at Location 3.
INFO:SoCBusHandler:opensbi Region added at Origin: 0x40f00000, Size: 0x00080000, Mode:  RW, Cached:  True, Linker:  True.
INFO:SoCBusHandler:plic Region added at Origin: 0xf0c00000, Size: 0x00400000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:plic added as Bus Slave.
INFO:SoCBusHandler:clint Region added at Origin: 0xf0010000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:clint added as Bus Slave.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00010000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00010000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 1.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 2.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x04000000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoC:CSR Bridge csr added.
INFO:SoCBusHandler:csr Region added at Origin: 0xf0000000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:csr added as Bus Slave.
INFO:SoCCSRHandler:csr added as CSR Master.
INFO:SoCBusHandler:Interconnect: InterconnectShared (1 <-> 6).
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCCSRHandler:sdram CSR allocated at Location 1.
INFO:SoCCSRHandler:supervisor CSR allocated at Location 4.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Finalized SoC:
[... 253 lines cut ...]
	at scala.tools.nsc.symtab.classfile.ClassfileParser.$anonfun$parse$2(ClassfileParser.scala:161)
	at scala.tools.nsc.symtab.classfile.ClassfileParser.$anonfun$parse$1(ClassfileParser.scala:147)
	at scala.tools.nsc.symtab.classfile.ClassfileParser.parse(ClassfileParser.scala:130)
	at scala.tools.nsc.symtab.SymbolLoaders$ClassfileLoader.doComplete(SymbolLoaders.scala:343)
	at scala.tools.nsc.symtab.SymbolLoaders$SymbolLoader.complete(SymbolLoaders.scala:250)
	at scala.reflect.internal.Symbols$Symbol.completeInfo(Symbols.scala:1542)
	at scala.reflect.internal.Symbols$Symbol.info(Symbols.scala:1514)
	at scala.reflect.internal.Definitions.scala$reflect$internal$Definitions$$enterNewMethod(Definitions.scala:49)
	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus$lzycompute(Definitions.scala:1134)
	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus(Definitions.scala:1134)
	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods$lzycompute(Definitions.scala:1438)
	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods(Definitions.scala:1420)
	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode$lzycompute(Definitions.scala:1450)
	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode(Definitions.scala:1450)
	at scala.reflect.internal.Definitions$DefinitionsClass.init(Definitions.scala:1506)
	at scala.tools.nsc.Global$Run.<init>(Global.scala:1213)
	at sbt.internal.parser.SbtParser$.<init>(SbtParser.scala:141)
	at sbt.internal.parser.SbtParser$.<clinit>(SbtParser.scala)
	at sbt.internal.parser.SbtParserInit$$anon$2.run(SbtParser.scala:191)
[error] java.lang.NoClassDefFoundError: Could not initialize class sbt.internal.parser.SbtParser$
[error] Use 'last' for the full log.
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
```
I downloaded `sbt` via `sudo pacman -S sbt`.
Any idea?

Dolu1990, 2024-04-01:
Hi,

That is curious XD
in pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project/build.properties
Can you try to update that to 1.9.9 ?

roby2014, 2024-04-02:
Hello. Where is that file exactly located?

roby2014, 2024-04-02:
Found it: `/home/roby/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project/build.properties`. It was using `1.6.0` for some reason.. Works good now, would be great to document this but im not sure where..

Dolu1990, 2024-04-02:
> /home/roby/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project/build.properties

That is normal it was on 1.6.0
The issue is more that SBT, i realy don't know why and how, often break just by itself magicaly on some plateform, and so, need to force it using something recent XD

closed 2024-04-04 by roby2014

Dolu1990, 2024-04-09:
@roby2014 On which OS / OS Version are you ?

roby2014, 2024-04-14:
EndeavourOS Linux x86_64  6.8.2-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 28 Mar 2024 17:06:35 +0000 x86_64 GNU/Linux

ozel, 2025-08-01:
I just ran into this issue on Ubuntu 24.04 LTS, after installing the current SBT Ubuntu package repo (https://repo.scala-sbt.org/scalasbt/debian) as explained [here](https://www.scala-sbt.org/1.x/docs/Installing-sbt-on-Linux.html#Installing+sbt+on+Linux).

This got me:
```
sbt --version
sbt runner version: 1.11.3
```

As explained above, I had to update the build.properties to this latest version in order to finish the build completely. 
I also needed to manually wipe the pythondata-cpu-vexriscv-smp module from the site-packages folder of my OSS-CAD-SUITE installation, because I could not find out how to fully clean the failed sbt build instead (partial sbt builds with different versions seem to be problematic...):

```
cd litex/pythondata-cpu-vexriscv-smp
echo sbt.version=1.11.3 > pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project/build.properties
rm -rf /opt/oss-cad-suite/lib/python3.11/site-packages/pythondata_cpu_vexriscv_smp*
cd ..
./litex_setup.py --install
```

Thanks for this great demo repo, I love it!