litex-hub/linux-on-litex-vexriscv #381: Regenerate the cluster of the pythondata-cpu-vexriscv_smp

elialaz, 2024-05-08:
Hi everyone,

I have a dubs about the regeneration process of the Verilog of the VexRiscV core, in particular I trying to modify the core inside the pythondata-cpu-vexriscv_smp repository and adding a custom plugin inside it, my dubs are about how can I regenerate the Verilog of the core with my custom implementation inside and then basically run the standard workflow for the bitstream build and load inside this repository.

Just for reference in my case I modified the standard VexRiscvSmpClusterGen class to include my plugin so it will be included in the standard config used for the build.

Any clarification should be helpfully 
Thanks a lot

elialaz, 2024-05-09:
For everyone I manage to find how to regenerate, jest run the generate.py script inside the pythondata-cpu-vexriscv-smp

I think should be useful to add this to the README under the link to the sdb

Now i have a new problem, basically the eth0 interface on the genesys 2 board doesn't work (besides my custom Core)

Here the log:

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading images/Image to 0x40000000 (7531468 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (82.3KB/s).
[LITEX-TERM] Uploading images/rv32.dtb to 0x40ef0000 (3083 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (67.2KB/s).
[LITEX-TERM] Uploading images/rootfs.cpio to 0x41000000 (3781632 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (81.5KB/s).
[LITEX-TERM] Uploading images/opensbi.bin to 0x40f00000 (53640 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (70.0KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imasu
BOOT HART Features  : pmp,scounteren,mcounteren,time
BOOT HART PMP Count : 16
Firmware Base       : 0x40f00000
Firmware Size       : 124 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b109
[    0.000000] Linux version 5.14.0 (florent@panda) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2021.08-381-g279167ee8d) 10.3.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Tue Sep 21 12:57:31 CEST 2021
[    0.000000] earlycon: liteuart0 at I/O port 0x0 (options '')
[    0.000000] Malformed early option 'console'
[    0.000000] earlycon: liteuart0 at MMIO 0xf0001000 (options '')
[    0.000000] printk: bootconsole [liteuart0] enabled
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x1 Version=0x8
[    0.000000] SBI TIME extension detected
[    0.000000] SBI IPI extension detected
[    0.000000] SBI RFENCE extension detected
[... 130 lines cut ...]
          Interrupt:2

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@buildroot:~# ifdown eth0
ifdown: interface eth0 not configured
root@buildroot:~# ifup eth0
ifup: ignoring unknown interface eth0
root@buildroot:~#

I think is something related to the interrupts on the interface

Dolu1990, 2024-05-13:
Hi,

>  just run the generate.py script inside the pythondata-cpu-vexriscv-smp

the pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster*.v files are just there as pregenerated. If you ask a config which isn't generated, then litex will run the SpinalHDL generation.
So, in other words, if you tweek the VexRiscv core and want litex to use the updated design, you just need to delete all the VexRiscvLitexSmpCluster*.v  files.

> Ethernet not working

It may be related to a DTS/DTB missmatch ?
Did you used the DTS/DTB generated by litex ? or used some from other devboard ?

elialaz, 2024-05-13:
Hi @Dolu1990,

Firstly thanks to the reply, I modified the config from witch the pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster*.v files are generated for simplicity and a more plug and play approach.

For the Ethernet I tried both the litex generated one and another one made by me but I constantly obtaining the interrupt on the interface that makes the OS do not recognize it.
If you have suggestions I could try it but anyway I will make some test in the next days.

Another thing, I managed to generate a buildroot configuration with an integrated cpp compiler to easily test programs inside Linux, I found another issue regarding the update of the addressed in the .json file but I cannot manage to make it work by uploading directly with UART so I m trying to load the image on the sd card and boot from them, is the “standard” procedure or I need to make some adjustments to make it work?

Thanks in advance

Dolu1990, 2024-05-13:
> If you have suggestions

So far i got ethernet to work fine on Arty A7 and Digilent Nexys Video.
In buildroot i had to run the command : udhcpc -i eth0 in order to get it up.

>  found another issue regarding the update of the addressed in the .json file

What is it ?

> is the “standard” procedure or I need to make some adjustments to make it work?

So, when i debug with small images, i'm often directly loading stuff via JTAG (~800 KB/s)
Else i'm using sdcard or the ethernet interface directly. I'm kinda never using the uart interface, as it is very slow.

elialaz, 2024-05-16:
Hi @Dolu1990, seems that I was having an issue on my side on my pc, now I resolved everything.
Thanks for the advices!

Can I ask you a question, running some c exec in the linux buildroot seems that the csr register access are restricted for every register, you added some extra configuration parameters for this? 

I need to disable this protection to be able to access some basic register like cycle and so on.

closed 2024-05-16 by elialaz

Dolu1990, 2024-05-17:
In user mode, you should normaly have access to rdcycle and instret.

is it working ?

elialaz, 2024-05-17:
No but I found out why, by default the fullCsr variable is false so the register are initialized with CsrAccess.NONE
I just changed it and then they are accessible

Dolu1990, 2024-05-17:
Ahhh right, i understood the question in reverse, sorry XD

elialaz, 2024-05-17:
@Dolu1990 do not worry :)

In the next month when I will be more free I will fork the repository and add more stuff in the documentation, small things that are already mentioned in the resolved issues and so on that maybe will help future users speed up their processes.

Dolu1990, 2024-05-17:
Thanks :D