litex-hub/linux-on-litex-vexriscv #382: Boot Crash Afer  --============= Liftoff! ===============--

hex55, 2024-05-11:
[image]

[image]

Why the boot halt ? How to fix it ? I use an EG4S20NG256B board, it same to sipeed tang primer, but it maked by ICMaker.

Dolu1990, 2024-05-13:
Hi,

The address used to load the binaries seems "not standard" related to the regular ones used in litex VexRiscv.
Did you updated the opensbi image ? (that would be the first step)

hex55, 2024-05-13:
Thanks Dolu1990! Because the board only has 8MB SDRAM, the address was changed like that. I haven't modified the opensbi code yet, but I didn't see any information output by opensbi. It seems that I didn't enter opensbi.

hex55, 2024-05-13:
> Hi,
> 
> The address used to load the binaries seems "not standard" related to the regular ones used in litex VexRiscv. Did you updated the opensbi image ? (that would be the first step)

Thanks Dolu1990! Because the board only has 8MB SDRAM, Jason's address was changed like that. I haven't modified the opensbi code yet, but I didn't see any information printed by opensbi. It seems that cpu didn't run opensbi. I will continue debugging tomorrow. Thank you for your help!

Dolu1990, 2024-05-13:
So, first step is to focus on opensbi only i would say, until this boot, nothing else realy mathers :)
But 8MB of sdram, i don't think you can run linux on this ? I mean, especialy if the filesystem is in ram.

hex55, 2024-05-14:
Hi 

> So, first step is to focus on opensbi only i would say, until this boot, nothing else realy mathers :) But 8MB of sdram, i don't think you can run linux on this ? I mean, especialy if the filesystem is in ram.

Hi Dolu1990, open sbi is work, next maybe as you save, I need put the rootfs in sd card. But I didn't find any information on this.
[image]

hex55, 2024-05-15:
> So, first step is to focus on opensbi only i would say, until this boot, nothing else realy mathers :) But 8MB of sdram, i don't think you can run linux on this ? I mean, especialy if the filesystem is in ram.

[image]

Dolu1990, 2024-05-15:
between ramfs and sdcard, there is some changes : 
- remove the cpio from the  boot.json 
- In the dts, bootargs need to have root=/dev/mmcblk0p2 instead, and no more linux,initrd-start linux,initrd-end
- linux driver for the sdcard  (they may already be in)
- dts need also to have the entry for the sdcard, but litex will already generate them in the dts normaly

hex55, 2024-05-16:
> between ramfs and sdcard, there is some changes :
>
> * remove the cpio from the  boot.json
> * In the dts, bootargs need to have root=/dev/mmcblk0p2 instead, and no more linux,initrd-start [...]

Thanks for your guidance, I've got it done.

closed 2024-05-16 by hex55